hypfvieh/bluez-dbus #3: When running on jdk8 and Raspbian, failed load library.

masato-ka, 2017-09-13:
Your library very helpful, thank you.

I will run the application on Raspberry Pi Zero W (Raspbian Jessei.)
The sample code compile is success . (Maven created runnable jar file.) But, When run the jar, An fatal error occurs on load native library.

My quation is "can the library running on below environment?".

 * Environment
    * JDK OpenJDK 8
    * OS   Raspbian Jessie lite

reference my sample code 
 * https://github.com/masato-ka/bluez-dbus-sample

Fatal error

~~~~
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  Internal Error (os_linux_zero.cpp:254), pid=1531, tid=3060204640
#  fatal error: caught unhandled signal 4
#
# JRE version: OpenJDK Runtime Environment (8.0_40-b04) (build 1.8.0_40-internal-b04)
# Java VM: OpenJDK Zero VM (25.40-b08 interpreted mode linux-arm )
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/bluez-dbus-sample/target/hs_err_pid1531.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted
~~~~

If you needs hs_err_pid.log, I upload that.

Best Regards.

hypfvieh, 2017-09-13:
Bluez-dbus requires libmatthew, which you have to install using apt. 
The library itself will work on raspberry. I use it myself.

masato-ka, 2017-09-13:
Thank you for your response.

I already installed libmatthew-io-java with below command.

~~~~
$sudo apt-cache search libmatthew-io-java
~~~~

result is below

~~~~
pi@raspberrypi:~ $ dpkg -L libmatthew-io-java
/.
/usr
/usr/share
/usr/share/java
/usr/share/java/io-0.1.jar
/usr/share/doc
/usr/share/doc/libmatthew-io-java
/usr/share/doc/libmatthew-io-java/copyright
/usr/share/doc/libmatthew-io-java/changelog.Debian.gz
/usr/share/doc/libmatthew-io-java/changelog.gz
/usr/share/java/io.jar
~~~~

However I can not find libunix-java.so in my raspbian.
I will check the library is  installing completely again.

Thank you.

hypfvieh, 2017-09-14:
I think libmatthew-io-java is not the correct library.
Afaik there are multiple libraries which are part of the libmatthew stuff.

If I remember correctly the package is called libmatthew-java (or maybe libunixsocket-java).
Sorry, I cannot look on my raspberry atm... :(

masato-ka, 2017-09-16:
I installed libunixsocket-java. but result is not change.
And i change JDK to oracle-idk from open-idk, but same it...
but I get another error message.
~~~~
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xac624e18, pid=2137, tid=3057173600
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# C  [libunix-java.so+0xe18]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/bluez-dbus-sample/hs_err_pid2137.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted
~~~~

This error is not unsatisfied library exception. I guess your library load own libunix-java.so that in matthew.jar(https://github.com/hypfvieh/libmatthew).
May be The library is not running on raspberry pi zero w. Because fatal error is SIGILL at libunix-java.so

OK, I will try to running libunix-java on raspberry Pi zero w.
If you have any information please tell me.

 Thank you.

masato-ka, 2017-09-16:
Good news
 I copy libunix-java.so to [execution-path]/lib and it work.

When library load from system, It seems to be not loading from LD_LIBRARY_PATH or PATH. At line 113 in NativeLibraryLoad::findProperNativeLib, _trySystemLibToo is always false this time. And , matthew.jar have library then the library is always read from the resource folder in matthew.jar.

Now library load priority.
1. Specified search path
2. classpath 
3. LD_LIBRARY_PATH

I suggestion that change the priority below to
1. Specified path
2. LD_LIBRARY_PATH
3. classpath

Thank you.

hypfvieh, 2017-09-18:
I will take a look at that, when I have some spare time. 
Thanks for working this out

masato-ka, 2017-09-18:
Thanks for your comment.
I close this issue. 

Best Regards.

closed 2017-09-18 by masato-ka