hypfvieh/bluez-dbus #6: Small Example to expose and consume a service...

Toomy, 2018-02-04:
Hi
Thank you for this Library. I try to understand how it works but I'm struggling a bit since it has no examples and I'm not exactly an BT expert.
Can you please provide an example on how to expose an consume simple Service, for instance a Temp service? That would be great.
Thank you Markus

hypfvieh, 2018-02-04:
Sorry, I don't have any examples.

Maybe you get an idea on how to use this library if you look at my usage of this library:
[PaulmannDeviceController](https://github.com/hypfvieh/paulmann-bt-lib/blob/master/src/main/java/com/github/hypfvieh/PaulmannDeviceController.java
)

hypfvieh, 2019-07-04:
Additional samples might be found here:
https://github.com/hypfvieh/sandbox/tree/master/src/main/java/com/github/hypfvieh/sandbox/bluez

closed 2019-07-04 by hypfvieh