hypfvieh/bluez-dbus #8: BluetoothDevice.getRssi() throwing during normal operation

falkreon, 2018-07-07:
bluez-dbus version: 0.0.2
dbus-java version: 2.7.5

After obtaining a BluetoothDevice from BluetoothManager::scanForBluetoothDevices, calling device.getRssi() produces this error:

https://gist.github.com/falkreon/661dfb564eb308c405f8ade69593f811

Diving a little deeper, in the [blueZ docs](https://github.com/RadiusNetworks/bluez/blob/master/doc/device-api.txt), it looks like the property name is correct, but it's an optional property. Is there any way we could make this call safer?

falkreon, 2018-07-08:
Since a lot of work has been done since then, I pulled down the latest github version (4d6ba65 as of this writing), and now RSSI works, but the service UUIDs are now returning null for devices which have services. It's an optional property, but data is definitely going missing.

falkreon, 2018-07-08:
Since I had the projects cloned in, I did some more poking around, and discovered that dbus-java has switched over to ArrayList from Vector. ( https://github.com/hypfvieh/dbus-java/blob/master/src/main/java/org/freedesktop/dbus/Marshalling.java#L350 ) Well! Swapping the type out at ( https://github.com/hypfvieh/bluez-dbus/blob/master/src/main/java/com/github/hypfvieh/bluetooth/wrapper/BluetoothDevice.java#L205 ) from Vector to ArrayList brought LE service advertisements back. It's probably a similar issue with RSSI.

falkreon, 2018-07-08:
After building the latest BlueZ stack and doing some manual queries with dbus-send, I can confirm that the RSSI errors are coming directly from BlueZ, and in this circumstance throwing a DBusExecutionException is entirely appropriate. Renamed the issue to reflect the remaining problem. Will fork and fix for my purposes now, but if you're still around I'd be interested in putting a PR in to fix this.

hypfvieh, 2018-07-08:
PR is always welcome.
I'll take a look sometime next week as soon as you issued a PR.

hypfvieh, 2018-07-09:
I removed the usage of Vector and changed it to List, so it should be working with the latest dbus-java version.

falkreon, 2018-07-09:
Yay, fixed by eba4d5f
In other news, I didn't get the newest blueZ stack. Once I did, the RSSI problem vanished too.

closed 2018-07-09 by falkreon