hypfvieh/bluez-dbus #11: Deserialization error when getting manufacturer data

miketrewartha, 2019-01-03:
First of all, thanks for this library, it's proving extremely helpful to the project I'm working on right now!

I've got a Kotlin app running in a Docker container (running some flavor of Debian) hosted in balenaOS (an IoT device Linux OS intended for running Docker containers), so there are some layers of complexity here. That said, I can start a LE scan and see devices/advertisements, but I'm getting a `DBusExecutionException` when trying to read manufacturer data from an advertisement:

```
org.freedesktop.dbus.exceptions.DBusExecutionException: Wrong return type (failed to de-serialize correct types: sun.reflect.generics.reflectiveObjects.TypeVariableImpl cannot be cast to java.lang.reflect.ParameterizedType )
        at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:169)
        at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:226)
        at com.sun.proxy.$Proxy22.Get(Unknown Source)
        at com.github.hypfvieh.bluetooth.wrapper.AbstractBluetoothObject.getTyped(AbstractBluetoothObject.java:92)
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.getManufacturerData(BluetoothDevice.java:280)
        ...
```

Any idea what might be going on or where I can start to debug?

miketrewartha, 2019-01-03:
On a potentially related note, I usually get this when trying to get the BLE address of a device after a scan:
```
Exception in thread "BleDeviceScanner" org.freedesktop.dbus.errors.UnknownObject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at org.freedesktop.dbus.errors.Error.getException(Error.java:150)
        at org.freedesktop.dbus.errors.Error.throwException(Error.java:180)
        at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:162)
        at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:226)
        at com.sun.proxy.$Proxy22.Get(Unknown Source)
        at com.github.hypfvieh.bluetooth.wrapper.AbstractBluetoothObject.getTyped(AbstractBluetoothObject.java:92)
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.getAddress(BluetoothDevice.java:312)
        ...
```

hypfvieh, 2019-01-07:
First of all, which version of bluez are you using?
In my experience, older versions  (< 5.48) of bluez are not working so well with BLE devices. 

About your exception:
There is a bug in getManufacturerData method.
The call to getTyped uses Map.class but it should be DBusMap.class. I fixed it in the latest commit (but I can not test it at work (no BLE devices available)).

The failure of getAddress is something different. 
BLE sometimes seems to act a bit weird.

Sometimes it seems like a device could be discovered by a scan, but is not completely usable when trying to query any data from it. 
I noticed the same behavior when using tinyb and also sometimes saw this when using the bluez provided commandline utils.
I'm not sure why this is happening, but I'm pretty sure that this is caused by bluez or the BLE implementation in bluez.

daviessm, 2019-01-07:
I've had problems with `getAddress` too, always when the device had been discovered and had been used for a while. I was calling it when I logged updates once a minute and periodically `getAddress` would fail for  every device until BlueZ was restarted. I put it down to a BlueZ problem and the app now saves the address on connection instead.

miketrewartha, 2019-01-07:
@hypfvieh I've tried it with various versions, all above 5.43 since that's when manufacturer data support seems to have been added to BlueZ. I think the highest I tested was 5.50, which seems to be current. I was about to start source diving today on this to see if I could figure out what was wrong, so I'll try out this latest commit using `DBusMap` and let you know how it works. If it doesn't, I'll see what I can hunt down. Thanks!

@daviessm That's the approach I was going to use, too. My problem was definitely related to scanning twice (I'm doing it in a tail-recursive way to essentially be able to scan continuously, occasionally processing the advertisements I've seen), though. I haven't seen that issue when only scanning once.

That reminds me @hypfvieh, it would be fantastic if there was a way to scan with some sort of listener that received continuous advertisements/devices instead of having to scan for a certain period, then restart scanning. In Kotlin land, I could imagine this being a scan function that returns a `ReceiveChannel`, but in Java land, I think something like Streams might work. Just a future feature idea.

miketrewartha, 2019-01-07:
@hypfvieh No dice. The exception I'm getting seems to be occurring before any code referencing that `DBusMap` type is called, so I'll dig in and see if I can figure out what's going on.

hypfvieh, 2019-01-08:
too bad ... 
Have you tried to obtain the manufacturer data using the bluez commandline utils too?
I would try that before digging through the exception and/or DBus stuff. Maybe it is a bluez issue (had some of some when using bluez + BLE).

If you have a bug fix or improvement, please issue a pull request.

About your callback idea:
I'm not so sure if this is working. 
If you enable discovery mode on the bluetooth adapter, you may not do certain things like connecting to a specific bluetooth device - it may depend on the used bluetooth adapter. 

On my tests with the integrated bluetooth of my Dell laptop, I always have to stop discovery mode to send commands to any device.

Another thing is: I can't find any information about callbacks which will signal your code that new devices have appeared on DBus.
What the DeviceManager is doing right now is to set the BT adapter to discovery mode, wait some seconds (timeout interval)  and then use DBus introspection XML to find new devices registered to DBus. 

So if you want to have some sort of permanent discovery mode, you have to do at least the introspection XML parsing periodically. 

You can implement this yourself. Use `DeviceManager.getAdapter()` and call `startDiscovery()` on the retrieved adapter to enable discovery mode (returns true if discovery mode is enabled).
Then create a thread which will read the introspection XML. I've updated the code so you can call `DeviceManager.findBtDevicesByIntrospection(BtAdapter)` to do that.

Please keep in mind that you may have multithread issue if you do something like this, so proper synchronization is required!

miketrewartha, 2019-01-08:
Yeah, I can spin up `bluetoothctl`, then `scan on`, then `info F4:17:60:47:FC:ED` (which is a device I know) and get this, which looks correct:

```
[bluetooth]# info F4:17:60:47:FC:ED
Device F4:17:60:47:FC:ED (random)
        Alias: F4-17-60-47-FC-ED
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        ManufacturerData Key: 0x0059
        ManufacturerData Value:
  01 91 7c 18 20 48 ff ff 80 80 00 00 ff ff 00 00  ..|. H..........
  00 ff 00 80 ff                                   .....
        AdvertisingFlags:
  06                                               .
```

I'll keep looking and if I find the issue, I'll at least report back... maybe issue a PR, depending on how much time I get. I'm super crunched right now.

As far as scanning while connecting, yeah, I've had some issues with that in the past when developing for Android, so I always manually handle pausing scans during connections, then resuming them afterwards. I noticed that `startDiscovery()` function yesterday so I'll toy around with it.

miketrewartha, 2019-01-08:
Alright @hypfvieh, I think I found the issue... It looks like it's actually an issue in the dbus-java dependency, not this library. In the `Marshalling` class' `static Object deSerializeParameter(Object _parameter, Type _type, AbstractConnection _conn)` function, there's a check to see if `_parameter` is an instance of `DBusMap` and if it is, it's assumed that the value type is a `ParameterizedType`, so `_type` gets cast to that. However, in the case of manufacturer data, the value is a `byte[]`, which is obviously not a `ParameterizedType`, so the cast fails with a `ClassCastException` that bubbles up, is caught, and is reformatted into the `DBusExecutionException` I pasted in my original message. 

If we wrap the cast in a try/catch and catch the `ClassCastException`, then try to instead get the type parameters of the map from `_parameter` itself (instead of `_type`), everything works:

```java
static Object deSerializeParameter(Object _parameter, Type _type, AbstractConnection _conn) throws Exception {
        // Omitted some stuff up here...

         if (_parameter instanceof DBusMap) {
             LOGGER.trace("Deserializing a Map");
             DBusMap<?,?> dmap = (DBusMap<?,?>) _parameter;

             // Ditch this:
-            Type[] maptypes = ((ParameterizedType) _type).getActualTypeArguments();

             // Use this instead:
+            Type[] maptypes;
+            try {
+                maptypes = ((ParameterizedType) _type).getActualTypeArguments();
+            } catch (ClassCastException e) {
+                maptypes = _parameter.getClass().getTypeParameters();
+            }

             for (int i = 0; i < dmap.entries.length; i++) {
                 dmap.entries[i][0] = deSerializeParameter(dmap.entries[i][0], maptypes[0], _conn);
                 dmap.entries[i][1] = deSerializeParameter(dmap.entries[i][1], maptypes[1], _conn);
             }
         }

        // Omitted some stuff down here, too...
}
```

Overall, it seems like `Marshalling` just never handled primitive arrays correctly. That said, that class was pretty hard for me to follow with types flying everywhere haha, so maybe I'm reading it incorrectly. Either way, making the change above fixes my issue and I'm able to see correct manufacturer data! Does this all sound right? If so, I can issue a PR for dbus-java with the fix.

hypfvieh, 2019-01-09:
Interesting catch.. 
I'll take a look at that - no need for a patch/PR.
Maybe there are more issues like that especially if a parameter of a map or list is an array ...

hypfvieh, 2019-01-09:
I've now fixed the issue in dbus-java. It doesn't seem to be an issue when using lists of some sort. 
It only happens when using Maps containing arrays. 

Thanks for the report and your patch (I improved it a bit).

miketrewartha, 2019-01-09:
No prob! Yeah, my fix was just a simple "Does this fix it?" kind of check. I like the if/else more, too. Any idea on when you'll get a new version of dbus-java out and then get a corresponding new version of bluez-dbus out? I can keep using my locally patched copies for near-term development, but this will be a blocking issue for me as soon as I need to go to production with the code that depends on bluez-dbus.

closed 2019-01-09 by miketrewartha