hypfvieh/bluez-dbus #13: Pairing using PIN code?

weliem, 2019-05-29:
Hi, great library, thanks for all the work you put in. I am using it now for my first BLE project on a Linux machine. The peripheral I am trying to connect with is a bloodpressure meter that requires PIN-based pairing.

I managed to discover the device and connect to it and discover the services. But now I need to deal with the PIN-pairing.

I understand I need to create an Agent1 object and register it as the default agent with the AgentManager1. But since the library doesn't have nice Java classes for Agent1 and AgentManager1 I trying to implement them myself.  However, I am struggling to get it to work.

Here is what I did:
1. Created a BluetoothAgent class
2. In the BluetoothAgent class I call exportObject to create it on the Dbus. Call succeeds.
3. Created a BluetoothAgentManager class
4. Added a method to DeviceManager to get the default AgentManager. Call succeeds.
4. I tried calling RegisterAgent and passing the BluetoothAgent. 

However, registering the Agent fails. It says "RegisterAgent, method with signature 'os' does not exist". 

Any tips on how to do it? If you want I can share my full code...

hypfvieh, 2019-05-31:
Yes please show at least some of your code, so I can try to reproduce what you where trying

weliem, 2019-06-01:
Ok then. I created a fork and added the classes BluetoothAgent and BluetoothAgentManager.

I also added 2 methods to the DeviceManager to get the BluetoothAgentManager and to create a BluetoothAgent.

See https://github.com/weliem/bluez-dbus/commit/65b5d8bcc5e0096eb27dddf3a7236e77350d9118

After that in my application code I do just this:

            // Setup agent
            BluetoothAgentManager agentManager = deviceManager.getBluetoothAgentManager();
            BluetoothAgent agent = deviceManager.getBluetoothAgent();
            agentManager.registerAgent(agent, "NoInputNoOutput");

The last line produces an exception:

Exception in thread "main" org.freedesktop.dbus.errors.UnknownObject: Method "RegisterAgent" with signature "os" on interface "org.bluez.AgentManager1" doesn't exist
at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
at org.freedesktop.dbus.errors.Error.getException(Error.java:156)
at org.freedesktop.dbus.errors.Error.throwException(Error.java:186)
at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:164)
at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:228)
at com.sun.proxy.$Proxy6.RegisterAgent(Unknown Source)
at com.github.hypfvieh.bluetooth.wrapper.BluetoothAgentManager.registerAgent(BluetoothAgentManager.java:66)

However, if I use D-feet I see the RegisterAgent method with parameters Object, String. So I have no idea what is going wrong here....

hypfvieh, 2019-06-03:
The problem seems to be the `getBluetoothAgentManger()` method.
You have to use "/org/bluez" as path instead of "/org/bluez/AgentManager1". 

Anyway, you don't need to patch `DeviceManager` or use any of the classes around it (e.g. `DbusHelper`, `AbstractBluetoothObject`).
You just need the interfaces for bluez found in package `org.bluez` of bluez-dbus, the rest is plain dbus stuff.

I've created a small example in my sandbox repo which uses DBus + org.bluez interfaces, see [AgentManagerExample](https://github.com/hypfvieh/sandbox/blob/master/src/main/java/com/github/hypfvieh/sandbox/bluez/AgentManagerExample.java).

weliem, 2019-06-03:
Thanks! That fixed my crash. I am still new to using DBus....

I am also now able to call requestDefaultAgent succesfully. So now I (think) I should get called on RequestPinCode when I connect to a device that needs a PIN code. 

That doesn't happen though...I get:

org.freedesktop.dbus.exceptions.DBusExecutionException: Software caused connection abort
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.freedesktop.dbus.errors.Error.getException(Error.java:156)
	at org.freedesktop.dbus.errors.Error.throwException(Error.java:186)
	at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:164)
	at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:228)
	at com.sun.proxy.$Proxy7.Connect(Unknown Source)
	at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.connect(BluetoothDevice.java:412)

hypfvieh, 2019-06-03:
Please always share your code, otherwise it is impossible to reproduce or understand whats going on.

weliem, 2019-06-03:
Sure, I will try to make a minimal application that runs...

weliem, 2019-06-03:
Ok, so here is a test project:

https://github.com/weliem/bluez-test

I made sure it ONLY does pairing. The rest of the application I am making is quite extensive so that would make it hard to test this issue.

So what you can see if you run it is that registering the Agent works. I then connect to my device and it connects. Then DBUS is trying to call back my Agent class but it looks like it cannot find it....it seems it is trying to call 'DisplayPassphrase' but it can't find my Agent....I think it is because it was never created on the DBus. Isn't the exportObject call for that?

Here is the full output:
hello world
19:31:14.957 [main] DEBUG org.freedesktop.dbus.connections.Transport - Connecting to UNIX: {path=/var/run/dbus/system_bus_socket}
19:31:15.213 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Creating message with serial 1
19:31:15.214 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appending sig: yyyy data: [66, 1, 0, 1]
19:31:15.214 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appending sig: ua(yv) data: [1, [[1, [o, /org/freedesktop/DBus]], [6, [s, org.freedesktop.DBus]], [2, [s, org.freedesktop.DBus]], [3, [s, Hello]]]]
19:31:15.214 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appended body, type: null start: 128 end: 128 size: 0
19:31:15.222 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - marshalled size ([0, 0, 0, 0]): 000000 00 00 00 00                                            ....              

19:31:15.224 [DBus Sender Thread-1] DEBUG org.freedesktop.dbus.MessageWriter - <= MethodCall(0,1) { Path=>/org/freedesktop/DBus, Interface=>org.freedesktop.DBus, Member=>Hello, Destination=>org.freedesktop.DBus } { }
19:31:15.230 [DBusConnection] DEBUG org.freedesktop.dbus.MessageReader - => MethodReturn(1,1) { Reply Serial=>1, Destination=>:1.3129, Sender=>org.freedesktop.DBus, Signature=>s } { :1.3129 }
19:31:15.230 [DBusConnection] DEBUG org.freedesktop.dbus.connections.impl.DBusConnection - Handling incoming method return: MethodReturn(1,1) { Reply Serial=>1, Destination=>:1.3129, Sender=>org.freedesktop.DBus, Signature=>s } { :1.3129 }
19:31:15.231 [DBusConnection] DEBUG org.freedesktop.dbus.MessageReader - => DBusSignal [clazz=null]
Retrieving AgentManager1
19:31:15.231 [DBusConnection] DEBUG org.freedesktop.dbus.connections.impl.DBusConnection - Handling incoming signal:
19:31:15.241 [DBus Worker Thread-1] DEBUG org.freedesktop.dbus.messages.DBusSignal - Converting signal to type: class org.freedesktop.DBus$NameAcquired
19:31:15.242 [DBus Worker Thread-1] DEBUG org.freedesktop.dbus.messages.DBusSignal - Creating signal of type class org.freedesktop.DBus$NameAcquired with parameters [/org/freedesktop/DBus, :1.3129]
19:31:15.242 [DBus Worker Thread-1] DEBUG org.freedesktop.DBus$NameAcquired - Creating message with serial 2
19:31:15.242 [DBus Worker Thread-1] DEBUG org.freedesktop.DBus$NameAcquired - Appending sig: yyyy data: [66, 4, 0, 1]
19:31:15.242 [DBus Worker Thread-1] DEBUG org.freedesktop.DBus$NameAcquired - Appending sig: ua(yv) data: [3, [[1, [o, /org/freedesktop/DBus]], [2, [s, org.freedesktop.DBus]], [3, [s, NameAcquired]], [8, [g, s]]]]
19:31:15.258 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Creating message with serial 3
19:31:15.258 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appending sig: yyyy data: [66, 1, 0, 1]
19:31:15.258 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appending sig: ua(yv) data: [3, [[1, [o, /org/bluez]], [6, [s, org.bluez]], [2, [s, org.freedesktop.DBus.Introspectable]], [3, [s, Introspect]]]]
19:31:15.258 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - Appended body, type: null start: 136 end: 136 size: 0
19:31:15.259 [main] DEBUG org.freedesktop.dbus.messages.MethodCall - marshalled size ([0, 0, 0, 0]): 000000 00 00 00 00                                            ....              

19:31:15.260 [DBus Sender Thread-1] DEBUG org.freedesktop.dbus.MessageWriter - <= MethodCall(0,3) { Path=>/org/bluez, Interface=>org.freedesktop.DBus.Introspectable, Member=>Introspect, Destination=>org.bluez } { }
19:31:15.262 [DBusConnection] DEBUG org.freedesktop.dbus.MessageReader - => MethodReturn(1,603) { Reply Serial=>3, Destination=>:1.3129, Sender=>:1.3124, Signature=>s } { <!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node><interface name="org.freedesktop.DBus.Introspectable"><method name="Introspect"><arg name="xml" type="s" direction="out"/>
</method></interface><interface name="org.bluez.AgentManager1"><method name="RegisterAgent"><arg name="agent" type="o" direction="in"/>
<arg name="capability" type="s" direction="in"/>
</method><method name="UnregisterAgent"><arg name="agent" type="o" direction="in"/>
</method><method name="RequestDefaultAgent"><arg name="agent" type="o" direction="in"/>
</method></interface><interface name="org.bluez.ProfileManager1"><method name="RegisterProfile"><arg name="profile" type="o" direction="in"/>
<arg name="UUID" type="s" direction="in"/>
<arg name="options" type="a{sv}" direction="in"/>
</method><method name="UnregisterProfile"><arg name="profile" type="o" direction="in"/>
</method></interface><node name="hci0"/></node> }
19:31:15.262 [DBusConnection] DEBUG org.freedesktop.dbus.connections.impl.DBusConnection - Handling incoming method return: MethodReturn(1,603) { Reply Serial=>3, Destination=>:1.3129, Sender=>:1.3124, Signature= [... cut ...]
19:31:20.908 [DBusConnection] DEBUG org.freedesktop.dbus.errors.Error - Appending sig: ua(yv) data: [9, [[4, [s, org.freedesktop.dbus.errors.UnknownObject]], [5, [u, 611]], [6, [s, :1.3124]], [8, [g, s]]]]
19:31:20.908 [DBusConnection] DEBUG org.freedesktop.dbus.errors.Error - Appending sig: s data: [/ is not an object provided by this process.]
19:31:20.908 [DBus Sender Thread-1] DEBUG org.freedesktop.dbus.MessageWriter - <= Error(0,9) { Error Name=>org.freedesktop.dbus.errors.UnknownObject, Reply Serial=>611, Destination=>:1.3124, Signature=>s } { / is not an object provided by this process. }

weliem, 2019-06-03:
If I look at the log, it seems that creating the Agent1 object doesn't lead to any DBus calls. Shouldn't the agent object be created on the DBus? 

In my first try I had this in my code of BluetoothAgent:

        try {
            _dbusConnection.exportObject(_dbusPath, this);
        } catch (DBusException e) {
            e.printStackTrace();
        }

I was hoping to create the Agent1 object on the DBus with that. How can a callback otherwise ever get back to this Java object?  Perhaps that is why you see the error at the end of the log....? Because the object was never created on the DBUS?

hypfvieh, 2019-06-04:
Yes you have to export it.

I still don't understand why you stick to your patched DeviceManager stuff, it is not necessary at all.
Also you should be more careful about your code style. In your sample there are class names which are starting with lowercase and also there are no package names, which is bad practice too.

I've updated my sample sandbox code. It is working for me, tested with my smartphone.

When connecting to my phone, a InputBox-Dialog pops up and asks for the PIN code generated by my smartphone. 
After entering the code, my machine gets paired with my phone. Connection is then getting lost as the sample code has ended and the test application has been shut down.
If I enter a wrong code, pairing will be refused due to wrong PIN.

weliem, 2019-06-04:
Yep, got it working now! Thanks!

I know it is not the cleanest test code but it was just to demonstrate the issue.

My application is using all of your classes from the Bluez-dbus library so I am using the DeviceManger a lot in my code. I realize it is not strictly necessary but that is what I am using. I am learning now how you are wrapping DBus code....

So I am trying to implement the BluetoothAgent and BluetoothAgentManager wrapper classes so that I don't have to keep using DBus calls directly.

closed 2019-06-04 by weliem