hypfvieh/bluez-dbus #18: Repositories for using bluez-dbus on OSGi

s5uishida, 2019-07-09:
For your information.

I created the following repositories to use bluez-dbus on OSGi.
- [dbus-java-with-java-utils-osgi](https://github.com/s5uishida/dbus-java-with-java-utils-osgi)
- [bluez-dbus-osgi](https://github.com/s5uishida/bluez-dbus-osgi)

If there are any inconveniences for you, please let me know.
Thank you for your work and all the contributors.

hypfvieh, 2019-10-30:
Maybe we should merge your OSGI stuff into bluez-dbus as I did in dbus-java (so the latest release of dbus-java provides an OSGi bundle).

I would like to do it like in dbus-java:
- Turn this project into multi-module maven project
 - Parent: bluez-dbus-parent
- Add module for the current code base (bluez-dbus)
- Add a new module for OSGi stuff (bluez-dbus-osgi)

What do you think?

s5uishida, 2019-10-30:
It sounds good.

I imported the bluez-dbus source code into an Eclipse plugin project and built the OSGi bundle. The MANIFEST.MF file at that time is below.
```
https://github.com/s5uishida/bluez-dbus-osgi/blob/master/master-201910102015-7cbe45ab7b7de0a210130d96cf41e41070c0f195-patch/MANIFEST.MF
```
Sorry that it's not a maven project.

I'm glad if you may use it as a reference.

hypfvieh, 2019-10-30:
I've just commited the required changes. Maybe you can build it locally and try if it is working for you

s5uishida, 2019-10-30:
Thank you very much.
I will try it later.

closed 2020-03-24 by hypfvieh