hypfvieh/bluez-dbus #21: I can not get some of the GattServices and GattCharacteristics UUIDs well.

s5uishida, 2019-07-11:
I am using it with Raspberry Pi 3B (Raspbian Buster Lite OS ver 2019-06-20).
This OS has BlueZ 5.50 as standard.

When I obtained GattServices and GattCharacteristics from TI SensorTag CC2650, I could not get the following UUIDs for battery level.
```
Service:        0000180f-0000-1000-8000-00805f9b34fb
Characteristic: 00002a19-0000-1000-8000-00805f9b34fb
```
The result of the following Java code is
```
	device.refreshGattServices();

	for (BluetoothGattService service : device.getGattServices()) {
		if (LOG.isDebugEnabled()) {
			LOG.debug("[{}] {} service - {}", device.getAdapter().getDeviceName(), device.getAddress(), service.getUuid());
		}
		service.refreshGattCharacteristics();
		for (BluetoothGattCharacteristic characteristic : service.getGattCharacteristics()) {
			if (LOG.isDebugEnabled()) {
				LOG.debug("[{}] {} characteristic - {}", device.getAdapter().getDeviceName(), device.getAddress(), characteristic.getUuid());
			}
		}
	}
```
The output was as follows, and the UUIDs related to the battery level were not found.
```
[hci0] B0:B4:48:B9:92:86 service - 00001801-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 service - 0000180a-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a23-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a24-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a25-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a26-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a27-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a28-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a29-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a2a-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 00002a50-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 service - f000aa00-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa01-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa02-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa03-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000aa20-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa21-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa22-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa23-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000aa40-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa41-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa42-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa44-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000aa80-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa81-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa82-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa83-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000aa70-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa71-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa72-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa73-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - 0000ffe0-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 characteristic - 0000ffe1-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 service - f000aa64-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa65-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000aa66-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000ac00-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ac01-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ac02-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ac03-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 service - f000ccc0-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ccc1-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ccc2-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 characteristic - f000ccc3-0451-4000-b000-000000000000 
[... 48 lines cut ...]
handle: 0x0036, char properties: 0x0a, char value handle: 0x0037, uuid: f000aa42-0451-4000-b000-000000000000
handle: 0x0038, char properties: 0x0a, char value handle: 0x0039, uuid: f000aa44-0451-4000-b000-000000000000
handle: 0x003b, char properties: 0x12, char value handle: 0x003c, uuid: f000aa81-0451-4000-b000-000000000000
handle: 0x003e, char properties: 0x0a, char value handle: 0x003f, uuid: f000aa82-0451-4000-b000-000000000000
handle: 0x0040, char properties: 0x0a, char value handle: 0x0041, uuid: f000aa83-0451-4000-b000-000000000000
handle: 0x0043, char properties: 0x12, char value handle: 0x0044, uuid: f000aa71-0451-4000-b000-000000000000
handle: 0x0046, char properties: 0x0a, char value handle: 0x0047, uuid: f000aa72-0451-4000-b000-000000000000
handle: 0x0048, char properties: 0x0a, char value handle: 0x0049, uuid: f000aa73-0451-4000-b000-000000000000
handle: 0x004b, char properties: 0x10, char value handle: 0x004c, uuid: 0000ffe1-0000-1000-8000-00805f9b34fb
handle: 0x0050, char properties: 0x0a, char value handle: 0x0051, uuid: f000aa65-0451-4000-b000-000000000000
handle: 0x0052, char properties: 0x0a, char value handle: 0x0053, uuid: f000aa66-0451-4000-b000-000000000000
handle: 0x0055, char properties: 0x0a, char value handle: 0x0056, uuid: f000ac01-0451-4000-b000-000000000000
handle: 0x0057, char properties: 0x0a, char value handle: 0x0058, uuid: f000ac02-0451-4000-b000-000000000000
handle: 0x0059, char properties: 0x0a, char value handle: 0x005a, uuid: f000ac03-0451-4000-b000-000000000000
handle: 0x005c, char properties: 0x12, char value handle: 0x005d, uuid: f000ccc1-0451-4000-b000-000000000000
handle: 0x005f, char properties: 0x08, char value handle: 0x0060, uuid: f000ccc2-0451-4000-b000-000000000000
handle: 0x0061, char properties: 0x08, char value handle: 0x0062, uuid: f000ccc3-0451-4000-b000-000000000000
handle: 0x0064, char properties: 0x1c, char value handle: 0x0065, uuid: f000ffc1-0451-4000-b000-000000000000
handle: 0x0068, char properties: 0x1c, char value handle: 0x0069, uuid: f000ffc2-0451-4000-b000-000000000000
handle: 0x006c, char properties: 0x0c, char value handle: 0x006d, uuid: f000ffc3-0451-4000-b000-000000000000
handle: 0x006f, char properties: 0x12, char value handle: 0x0070, uuid: f000ffc4-0451-4000-b000-000000000000
[B0:B4:48:B9:92:86][LE]> 
```
If there is an error in the Java code using bluez-dbus, it would be very helpful for me if you could point it out.
Thank you very much.

hypfvieh, 2019-07-15:
I can't see any obvious mistakes.

As bluez-dbus is just querying DBus for registered objects, you may check if you can see the characteristics on DBus BEFORE using bluez gatttool or similar (e.g. using d-feet).

If they are not on DBus, then something has to be done before scanning the bus for objects. 
If it is present, you may have to debug the stuff happening in com.github.hypfvieh.bluetooth.wrapper.BluetoothGattCharacteristic#refreshGattCharacteristics()

s5uishida, 2019-07-15:
Thank you for your reply.

As additional information, the GATTService UUIDs obtained with `device.getUuids()` contained the GattService UUID including GattCharacteristic indicating the battery level as shown by the arrow below.

```
	for (String uuid : device.getUuids()) {
		LOG.info("[{}] {} uuid:{}", device.getAdapter().getDeviceName(), device.getAddress(), uuid);
	}
```
```
[hci0] B0:B4:48:B9:92:86 uuid:00001800-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 uuid:00001801-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 uuid:0000180a-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 uuid:0000180f-0000-1000-8000-00805f9b34fb <==
[hci0] B0:B4:48:B9:92:86 uuid:0000ffe0-0000-1000-8000-00805f9b34fb 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa00-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa20-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa40-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa64-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa70-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000aa80-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000ac00-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000ccc0-0451-4000-b000-000000000000 
[hci0] B0:B4:48:B9:92:86 uuid:f000ffc0-0451-4000-b000-000000000000
```
I will try to investigate the cause more.

s5uishida, 2019-07-16:
I was able to get the battery level.

In `DbusHelper#findNodes()` called from `BluetoothDevice#refreshGattServices()`, for Device1 object,
```
Introspectable remoteObject = _connection.getRemoteObject("org.bluez", _path, Introspectable.class);
String introspect = remoteObject.Introspect();
```
The introspect text looks like this. I made Line breaks for easy viewing.
As an arrow, the `org.bluez.Battery1` interface existed.
```
<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN" "http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node>

<interface name="org.freedesktop.DBus.Introspectable">
    <method name="Introspect">
        <arg name="xml" type="s" direction="out"/>
    </method>
</interface>

<interface name="org.bluez.Device1">
    <method name="Disconnect"></method>
    <method name="Connect"></method>
    <method name="ConnectProfile">
        <arg name="UUID" type="s" direction="in"/>
    </method>
    <method name="DisconnectProfile">
        <arg name="UUID" type="s" direction="in"/>
    </method>
    <method name="Pair"></method>
    <method name="CancelPairing"></method>
    <property name="Address" type="s" access="read"></property>
    <property name="AddressType" type="s" access="read"></property>
    <property name="Name" type="s" access="read"></property>
    <property name="Alias" type="s" access="readwrite"></property>
    <property name="Class" type="u" access="read"></property>
    <property name="Appearance" type="q" access="read"></property>
    <property name="Icon" type="s" access="read"></property>
    <property name="Paired" type="b" access="read"></property>
    <property name="Trusted" type="b" access="readwrite"></property>
    <property name="Blocked" type="b" access="readwrite"></property>
    <property name="LegacyPairing" type="b" access="read"></property>
    <property name="RSSI" type="n" access="read"></property>
    <property name="Connected" type="b" access="read"></property>
    <property name="UUIDs" type="as" access="read"></property>
    <property name="Modalias" type="s" access="read"></property>
    <property name="Adapter" type="o" access="read"></property>
    <property name="ManufacturerData" type="a{qv}" access="read"></property>
    <property name="ServiceData" type="a{sv}" access="read"></property>
    <property name="TxPower" type="n" access="read"></property>
    <property name="ServicesResolved" type="b" access="read"></property>
</interface>

<interface name="org.freedesktop.DBus.Properties">
    <method name="Get">
        <arg name="interface" type="s" direction="in"/>
        <arg name="name" type="s" direction="in"/>
        <arg name="value" type="v" direction="out"/>
    </method>
    <method name="Set">
        <arg name="interface" type="s" direction="in"/>
        <arg name="name" type="s" direction="in"/>
        <arg name="value" type="v" direction="in"/>
    </method>
    <method name="GetAll">
        <arg name="interface" type="s" direction="in"/>
        <arg name="properties" type="a{sv}" direction="out"/>
    </method>
    <signal name="PropertiesChanged">
        <arg name="interface" type="s"/>
        <arg name="changed_properties" type="a{sv}"/>
[... 92 lines cut ...]
 * Bluetooth device ObjectType identifier.
 * @author hypfvieh
 *
 */
public enum BluetoothDeviceType {
    NONE, ADAPTER, DEVICE, GATT_SERVICE, GATT_CHARACTERISTIC, GATT_DESCRIPTOR, BATTERY
}
```
With this, I could get the battery level with the following code.
```
Battery1 rawBattery = DbusHelper.getRemoteObject(manager.getDbusConnection(), device.getDbusPath(), Battery1.class);
BluetoothBattery battery = new BluetoothBattery(rawBattery, device, device.getDbusPath(), manager.getDbusConnection());

device.connect();

LOG.info("battery level:{}", (int)battery.getPercentage())
```
Another one, I am creating `CC2650Driver.java`.
In the code to get the battery level, I need to call `manager.getDbusConnection()`, so the constructor is as follows:
```
public CC2650Driver(DeviceManager manager, BluetoothDevice device)
```
However, when AbstractBluetoothObject#getDbusConnection() goes from protected to public, the first argument `manager` becomes unnecessary and easy.

I understand that you have a policy on the scope of `method`.

s5uishida, 2019-07-16:
It would be very pleasure for me to commit these to bluez-dbus.

hypfvieh, 2019-07-16:
Nice catch.
The documentation of Battery1 is also available in the official bluez git found at:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/battery-api.txt

If you want to contribute your additions, please create a pull request.

s5uishida, 2019-07-16:
Thank you very much.

With this I could get the following information from CC2650.
- Firmware version
- Battery level
- Sensor data
	- Temperature(Object/Ambience)
	- Humidity
	- Barometric pressure
	- Illuminance
	- Movement
		- Gyroscope (x-y-z)
		- Accelerometer (x-y-z)
		- Magnetometer (x-y-z)

I could get these data in two ways: direct reading and notification.

In addition, in the case of CC2650, the maximum interval is 2550 (msec) in the notification method, so it is not suitable for such as one minute interval in practice.

So I will use direct read method.

This issue was solved with alternatives (Battery1 I/F).

Thank you very much again.

closed 2019-08-01 by hypfvieh