hypfvieh/bluez-dbus #27: Order of notifications is not always correct at high speed

weliem, 2019-09-18:
Hi,

I have a weird issue I don't understand. I am working with a Pulse Oximeter and an Activity Tracker that stream data at high speed to my application. Most of the times the order of notifications is correct but sometimes the packets are in the wrong order. So it seems packets are not guaranteed to arrive in the order they were sent by the devices. This never happens on iOS or Android...

When I see this issue happening my propertyChanged handler gets called 2x in less than 1ms. If there is more than 1ms between the calls everything seems fine and the order is correct.

Any idea where the issue could be? Is it Bluez itself? Or de Java DBus library? Or any idea how to debug this?

Here is my code for handling notifications:

```
    private AbstractPropertiesChangedHandler characteristicUpdatesHandler = new AbstractPropertiesChangedHandler() {
        @Override
        public void handle(Properties.PropertiesChanged propertiesChanged) {
            // Check if it is for this device
            String deviceAddressString = deviceAddress.replace(":","_");
            if(!propertiesChanged.getPath().contains(deviceAddressString)) return;

            if (propertiesChanged.getInterfaceName().equals("org.bluez.GattCharacteristic1")) {
                String path = propertiesChanged.getPath();

                propertiesChanged.getPropertiesChanged().forEach((s, value) -> {
                    // See what the update is about
                    if (s.equalsIgnoreCase("Notifying")) {
                        boolean isNotifying = (Boolean) value.getValue();
                        hbDevice.onNotifySet(getCharacteristicFromPath(path), isNotifying );
                    } else if (s.equalsIgnoreCase("Value")) {
                        if (value.getType() instanceof DBusListType) {
                            if (value.getValue() instanceof byte[]) {
                                byte[] byteVal = (byte[]) value.getValue();
                                byte[] valueCopy = byteVal.clone();
                                HBCharacteristic hbCharacteristic = getCharacteristicFromPath(path);
                                hbCharacteristic.setValue(valueCopy);
                                hbDevice.onCharacteristicChanged(valueCopy, hbCharacteristic);
                            }
                        }
                    }
                });
            }
        }
    };
```

hypfvieh, 2019-09-19:
Pretty difficult to find who is responsible for this behavior.

DBus usually preserves the order of the messages in the order it has been written to the bus.
So maybe bluez is messing up.

Did you try to reproduce the issue using some other bluez-based tool, like the provided 'bluetoothctl' command?
If that tool also receives messages in the wrong order, it is be a bluez issue.

If the test above does not show the same results as you see in Java I would bet on the dbus-java library.

The code of dbus-java has been improved by me and some others, but it is still based on a very old implementation. 

Converting DBus messages from/to Java requires a lot of String comparison, switch cases, ifs and loops. 
I don't think that this is very fast, as String stuff in Java is usually slower than working with primitives. So maybe the bottle neck is in that code.

weliem, 2019-09-28:
Thanks for the suggestions! Reproducing with bluetoothctl is practically hard because these devices have a proprietary protocol where I would have to issue more than 20 command before I get to the state where I need to be....but maybe I should try anyway...

I was thinking of first monitoring the lowlevel traffic using **btmon**. Hopefully I can then see if the packages arrive in the right order....after that I need to see if it is the dbus-java code.

weliem, 2020-08-26:
Ok, sorry for the late reply. I finally had some time to debug this.

First thing I did was a using **btmon** to see what that showed and that was looking all fine. The notifications are coming in in the correct order.

Next I used the dbus-monitor to see if the signals are correctly timestamped. And yes, they are:
```
signal time=1598444028.981149 sender=:1.252 -> destination=(null destination) serial=7923741 path=/org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               77 1e 06 ee 04 00 00 04 00 00 00 15 1a 46 5f 64 64 64 64 ff
            ]
      )
   ]
   array [
   ]
signal time=1598444028.982240 sender=:1.252 -> destination=(null destination) serial=7923742 path=/org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               ff ff ff ff ff ff ff ff ff ff ff b9
            ]
      )
   ]
   array [
   ]
```

But in my application I received these two notifications in the reversed order:
```
14:13:48.997 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <FFFFFFFFFFFFFFFFFFFFFFB9>
14:13:48.997 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Got unexpected data <FFFFFFFFFFFFFFFFFFFFFFB9>
14:13:48.998 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <771E06EE04000004000000151A465F64646464FF>
```

As you can see in the timestamps, the signals come in **very fast** after each other. So my current hypothesis is that the order is perhaps messed up somewhere in dbus-java. Perhaps an issue with the thread pool that handles the signals?

One thing I still want to do is to turn on debug mode so that I can also see what happens inside dbus-java.....will send that soon.

weliem, 2020-08-26:
Ok, now with debug on I see this:

```
15:17:46.684 [DBusConnection] DEBUG o.f.d.c.impl.DBusConnection - Handling incoming signal: DBusSignal [clazz=null]
15:17:46.684 [DBus Worker Thread-1] DEBUG o.f.dbus.messages.DBusSignal - Converting signal to type: class org.freedesktop.dbus.interfaces.Properties$PropertiesChanged
15:17:46.685 [DBus Worker Thread-2] DEBUG o.f.d.i.Properties$PropertiesChanged - Creating message with serial 402
15:17:46.685 [DBus Worker Thread-1] DEBUG o.f.d.i.Properties$PropertiesChanged - Creating message with serial 403
15:17:46.685 [DBus Worker Thread-1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: yyyy data: [108, 4, 0, 1]
15:17:46.685 [DBus Worker Thread-1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: ua(yv) data: [404, [[1, [o, /org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, PropertiesChanged]], [8, [g, sa{sv}as]]]]
15:17:46.685 [DBus Worker Thread-2] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: yyyy data: [108, 4, 0, 1]
15:17:46.685 [DBus Worker Thread-2] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: ua(yv) data: [403, [[1, [o, /org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, PropertiesChanged]], [8, [g, sa{sv}as]]]]
15:17:46.689 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <0056>
15:17:46.689 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Got unexpected data <0056>
15:17:46.689 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <77140128101F00030100170010E5040000F50400>
```
So it looks like the reversing indeed happens in dbus-java. Message with serial 404 is delivered before message 403....

weliem, 2020-08-26:
I suspect the issue is that since each signal is handled by a different thread, one is processed faster than the other and hence the order is messed up....

hypfvieh, 2020-08-27:
I think you found the issue, well done.

For your current issue you may try to change the thread pool size (calling `connection.changeThreadCount(1)`).
This will change the working thread pool size to 1 thread.

It will be pretty hard to get this one fixed properly. 
The idea behind the thread pool was to handle multiple (different) signals at the same time without blocking other incoming signals.
This may happen if one creates a signal handler which is 'slow' in some matter. It would then stop processing all other signals until the handler is done. That can cause other issues like messages get lost, overflow of connection buffer etc.

Guarantee the order would be pretty hard, because what would be the 'good way' to behave when receiving msgNo 10 and the last 'known' number was 8? 
Waiting until 9 has also arrived? 
For how long? 
What if msgNo 9 never arrives?

I'm not sure that  there will be 'one' way to solve it all. Like said above, using one thread can cause blocking issues, using multiple may mixup the order... 

One possible solution would be to read all messages with one thread and put them in a queue. The queue will be read by a different thread which than will execute the signal handlers provided. This would fix the problem with loosing messages if message handler is too slow, but will also slow down message processing.

Another option may be to try to execute certain signals with the same thread, like adding a thread for each signal handler.
But I think this could cause thread leaking, because you never know when a thread is now longer required.

'Pinning' threads to certain signal handlers using a thread pool is also difficult. To know which thread/handler to use, the message has to be parsed to a certain point and I guess that it's already too late to decide which thread to use.

weliem, 2020-08-27:
Yes I agree....tricky to solve...

I quickly tried your suggestion to use changeThreadCount. It makes the problem occur less often but it is still not entirely gone:

```
10:46:24.849 [DBusConnection] DEBUG o.f.d.c.impl.DBusConnection - Handling incoming signal: DBusSignal [clazz=null]
10:46:24.849 [DbusWorkerThreads1] DEBUG o.f.dbus.messages.DBusSignal - Converting signal to type: class org.freedesktop.dbus.interfaces.Properties$PropertiesChanged
10:46:24.849 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Creating message with serial 318
10:46:24.849 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: yyyy data: [108, 4, 0, 1]
10:46:24.849 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: ua(yv) data: [319, [[1, [o, /org/bluez/hci0/dev_60_E0_E0_27_2B_7E]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, PropertiesChanged]], [8, [g, sa{sv}as]]]]
10:46:24.852 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Creating message with serial 319
10:46:24.852 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Appending sig: yyyy data: [108, 1, 0, 1]
10:46:24.852 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Appending arguments with signature: ss
10:46:24.852 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Appending sig: ua(yv) data: [319, [[1, [o, /org/bluez/hci0/dev_60_E0_E0_27_2B_7E]], [6, [s, org.bluez]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, Get]], [8, [g, ss]]]]
10:46:24.852 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Appending sig: ss data: [org.bluez.Device1, Address]
10:46:24.853 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - Appended body, type: ss start: 152 end: 188 size: 36
10:46:24.853 [Looper-run-CentralQueue-signal] DEBUG o.f.dbus.messages.MethodCall - marshalled size ([36, 0, 0, 0]): 000000	24 00 00 00                                            $...              
10:46:24.853 [DBus Sender Thread-1] DEBUG o.f.d.spi.OutputStreamMessageWriter - <= MethodCall(0,319) { Path=>/org/bluez/hci0/dev_60_E0_E0_27_2B_7E, Interface=>org.freedesktop.DBus.Properties, Member=>Get, Destination=>org.bluez, Signature=>ss } { org.bluez.Device1, Address }
10:46:24.854 [DBusConnection] DEBUG o.f.d.spi.InputStreamMessageReader - => MethodReturn(1,22269) { Reply Serial=>319, Destination=>:1.138, Sender=>:1.3, Signature=>v } { [60:E0:E0:27:2B:7E] }
10:46:24.854 [DBusConnection] DEBUG o.f.d.c.impl.DBusConnection - Handling incoming method return: MethodReturn(1,22269) { Reply Serial=>319, Destination=>:1.138, Sender=>:1.3, Signature=>v } { [60:E0:E0:27:2B:7E] }
10:46:24.930 [DBusConnection] DEBUG o.f.d.spi.InputStreamMessageReader - => DBusSignal [clazz=null]
10:46:24.930 [DBusConnection] DEBUG o.f.d.c.impl.DBusConnection - Handling incoming signal: DBusSignal [clazz=null]
10:46:24.932 [DbusWorkerThreads1] DEBUG o.f.dbus.messages.DBusSignal - Converting signal to type: class org.freedesktop.dbus.interfaces.Properties$PropertiesChanged
10:46:24.932 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Creating message with serial 320
10:46:24.932 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: yyyy data: [108, 4, 0, 1]
10:46:24.933 [DBusConnection] DEBUG o.f.d.spi.InputStreamMessageReader - => DBusSignal [clazz=null]
10:46:24.933 [DBusConnection] DEBUG o.f.d.c.impl.DBusConnection - Handling incoming signal: DBusSignal [clazz=null]
10:46:24.933 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: ua(yv) data: [321, [[1, [o, /org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, PropertiesChanged]], [8, [g, sa{sv}as]]]]
10:46:24.936 [DbusWorkerThreads1] DEBUG o.f.dbus.messages.DBusSignal - Converting signal to type: class org.freedesktop.dbus.interfaces.Properties$PropertiesChanged
10:46:24.936 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Creating message with serial 321
10:46:24.936 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: yyyy data: [108, 4, 0, 1]
10:46:24.937 [DbusWorkerThreads1] DEBUG o.f.d.i.Properties$PropertiesChanged - Appending sig: ua(yv) data: [322, [[1, [o, /org/bluez/hci0/dev_C5_E2_E1_19_9A_BD/service0013/char0016]], [2, [s, org.freedesktop.DBus.Properties]], [3, [s, PropertiesChanged]], [8, [g, sa{sv}as]]]]
10:46:24.938 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <403E40FFFFFFFFFFFFFFFF92>
10:46:24.938 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Got unexpected data <403E40FFFFFFFFFFFFFFFF92>
10:46:24.939 [Looper-run-Central-callBackHandler] INFO   - MasimoPulseOximeterServiceHandler: Received data <771E06090500000B00000000000000606060603D>
```

Do you think there might be another issue that is also relevant? Not sure how to read the log....the message serial is increasing correctly but I still receive the out of order...

hypfvieh, 2020-08-27:
Hmm....... 

With the thread pool size of 1, dbus-java uses 2 threads to handle everything:
One thread (IncomingMessageThread, not part of the thread-pool) will poll for messages on the bus which needs to be handled (logged as [DBusConnection] thread). 
The second thread will be executed when the message handler has been determined and the message/signal is going to be processed (part of the thread pool, logged as [DbusWorkerThreads1]).
As mention before, this is to avoid blocking everything when the message handler is slow.

For me it does not look like the order will be changed here. 
If there are multiple responses on the bus which needs to be handled then it will add additional runnables to the thread-pool which will be handled as soon as the only thread in the pool is available again.

Do you use any kind of (Hash)Set or (Hash)Map in your code which may mess up the order?

weliem, 2020-08-27:
Ok.....I'll dig a bit more into this. I am also offloading incoming propertiesChanged events to a Thread with a queue. I'll add some more logging before handing it to the Thread. It could indeed be my code...

weliem, 2020-08-27:
I have good news! There was indeed also a problem in my code.

Quite interesting issue. The problem was that I put the propertiesChanged object on a message queue and the objects are handled based on the timestamp they have. But I was only comparing timestamps in milliseconds resolution. But sometimes the objects would come in so fast that they would have the same timestamp in milliseconds because they came in less than 1 ms apart. The consequence was that the order of the two objects would then be reversed. So now I updated my code to look also at the nanoseconds part of the timestamp and everything works flawlessly!

Anyway, my device now works properly so the problem is solved. Will close this issue, although I still think it would be nice if dbus-java could be used with a thread pool while also giving the guarantee that propertiesChanged objects are always in the right order. Perhaps something for on your dbus-java roadmap?

closed 2020-08-27 by weliem