hypfvieh/bluez-dbus #28: bluez-dbus works with dbus-java-parent-3.2.0?

s5uishida, 2019-10-30:
Hello.

Recently I knew the release of dbus-java-parent-3.2.0, which replaced libmatthew with jnr-unixsocket. It's great!

By the way, may bluez-dbus work with it? I think that I should try it myself, but if you have already tried it, please let me know that it works or not.

Thank you very much for your efforts.

hypfvieh, 2019-10-30:
Yes you can replace dbus-java 3.0.x with dbus-java 3.2.0.
There are no API breaking changes, the biggest change was to remove libmatthew, but that only changed some internals and does not cause issues with bluez-dbus.

I updated the current development version of bluez-dbus to use the newer dbus-java version instead of the 3.0.2-SNAPSHOT.

s5uishida, 2019-10-30:
OK. I will try it.

Thank you very much.

s5uishida, 2019-11-02:
I confirmed the operation in the following cases.

1. **bluez-dbus-osgi (all-in-one bundle for bluez-dbus, dbus-java, jnr-unixsocket and etc...)**
  The bluez-dbus-osgi bundle includes jnr-unixsocket, so an import declaration for sun.misc was required.

1. **dbus-java-osgi (dbus-java, jnr-unixsocket and etc...) + bluez-dbus-osgi (no embedded jars)**
  Since dbus-java-osgi includes jnr-unixsocket, an import declaration for sun.misc was required. On the other hand, when creating a bluez-dbus-osgi bundle without embedded jars such as jnr-unixsocket, the import declaration of sun.misc was unnecessary.

I think the later method is good.

The reason is that dbus-java-osgi includes all set for using dbus in java, and for application protocols on dbus such as the bluez, a small bundle without embedded jars is enough.

What do you think?

hypfvieh, 2019-11-02:
I'm fine with both versions... I don't use OSGi and don't have any experience with it.
The pom for dbus-java was provided by some contributor and I just added it as maven module.

s5uishida, 2019-11-03:
OK. I will use the latter method for handling each update of dbus-java and bluez-dbus.

Thank you very much.

closed 2020-03-24 by hypfvieh