hypfvieh/bluez-dbus #35: GATT flags are returned as a String rather than an array of Strings

srware, 2020-06-29:
Currently the getFlags() method of BluetoothGattDescriptor and BluetoothGattCharacteristic returns a String. I would expect the translation from DBus API to Java in this case to be String[] given it returns an array of strings.

https://github.com/hypfvieh/bluez-dbus/blob/22bc2ba9c3679a2bdd8880e307f421ad9482090d/bluez-dbus/src/main/java/com/github/hypfvieh/bluetooth/wrapper/BluetoothGattDescriptor.java#L143
https://github.com/hypfvieh/bluez-dbus/blob/22bc2ba9c3679a2bdd8880e307f421ad9482090d/bluez-dbus/src/main/java/com/github/hypfvieh/bluetooth/wrapper/BluetoothGattCharacteristic.java#L213

Is this a bug or does the end user need to do a contains() to determine which flags are present?

hypfvieh, 2020-06-30:
I think this is a bug. 
The bluez docs say that 'flags' are an array of String, so the method should query and return String[] or List<String>.

I changed that in the latest commit, maybe you can try that. I don't have any device to test this right now.

srware, 2020-06-30:
@hypfvieh , thanks for the fast response. I am also struggling to find something that returns a characteristic with properties defined but in terms of implementation the fix works so in principal hopefully this is fixed :)

closed 2020-06-30 by srware