hypfvieh/bluez-dbus #38: Adapter isDiscovering() behaviour

srware, 2020-07-09:
I am seeing some strange behaviour with the isDiscovering() call in BluetoothAdapter.

I am frequently getting the following error when calling stopDiscovery() despite the isDiscovering() function returning "true":

`org.freedesktop.dbus.exceptions.DBusExecutionException: No discovery started`

I guess this exception should also be caught in the BluetoothAdapter.stopDiscovery() function. I am also wondering whether initialising the `internalDiscover` variable to false would help this although I guess that assumption could lead to other problems...

hypfvieh, 2020-07-10:
The member `internalDiscover` is only used when `startDiscovery()` / `stopDiscovery()` is used. 
The method `isDiscovering()` will return true if `internalDiscovery` is true (when `startDiscovery() `was called and `stopDiscovery() `was not yet called), or if the DBus-Response says discovery is active.

The internal member does not need to be initialized with false, because that is the default for primitive boolean members in Java.

The exception might occur because `internalDiscovery` was not set correctly when `startDiscovery()` / `stopDiscovery()` was called due to exception catched in these methods. Therefore `isDiscovering()` still returns true/false and the start/stop method will be executed instead of being omitted.

Setting the member and calling start/stopDiscovery on DBus is not atomic, so in the current code, DBus is called first, then the member is changed. If DBus throws an exception, setting the member to true/false is never called, so the `internalDiscovery` flag has the wrong state.

I change the code to first set the member, then call DBus.

srware, 2020-07-10:
Thanks for the explanation @hypfvieh 

Is there a timeline for a new release incorporating some of the recent fixes?

hypfvieh, 2020-07-10:
no there is no timeline

closed 2020-07-10 by srware