hypfvieh/bluez-dbus #39: Help is more like it. I am not able to see my ble device during `scanForBluetoothDevices` period

abunaiquad, 2020-07-25:
Hi,

I am very new to bluetooth so please bear with me. 

I can only get 1 ble device in `scanForBluetoothDevices` list and it wasn't my device. I can see my ble device in `btmon` log during `scanForBluetoothDevices` period. Could you please provide some info to help me move further?

Thanks much.

hypfvieh, 2020-07-26:
If you need help, it would be great if you could share your code. Otherwise I can only guess what's going on.

abunaiquad, 2020-07-27:
> If you need help, it would be great if you could share your code. Otherwise I can only guess what's going on.

Thanks for responding!

I utilized codes from s5uishida

`var manager: DeviceManager = DeviceManager.createInstance(false)
        val filter: MutableMap<DiscoveryFilter, Any> = HashMap()
        filter[DiscoveryFilter.Transport] = DiscoveryTransport.LE
        filter[DiscoveryFilter.DuplicateData] = true
        manager.setScanFilter(filter)
        var selectedDevice: BluetoothDevice? = null
        log.info { "scanning..." }
        for (device in manager.scanForBluetoothDevices("hci0", 60000)) {
            val deviceName = device.name
            log.info("[{}] {} {}", device.adapter.deviceName, device.address, deviceName)
            if (deviceName == "XYZ") {
                selectedDevice = device
                break
            }
        }
        checkNotNull(selectedDevice) { "device not found." }`

===============================================================================

When I set `sessionConnection` to true, I get different error. Could you please clarify what it means by sessionConnection true/false? 

`var manager: DeviceManager = DeviceManager.createInstance(true)`

`23:02:42.504 ERROR (com.github.hypfvieh.DbusHelper-59) Critical error while reading DBUS response (maybe no bluetoothd daemon running?) 
org.freedesktop.dbus.errors.ServiceUnknown: The name org.bluez was not provided by any .service files
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
`

hypfvieh, 2020-07-27:
First of all, it would be great if you could use some sort of code formatting. Reading the mess above is pretty hard.

So what can go wrong?
In your first case, you are using DiscoveryTransport-Filters. 
If you don't know what they are good for, don't use them. If you use filters and your bluetooth device does not match those filters, you won't see them.

Using `DeviceManager.createInstance(true)` will tell the DBus connection to use a session instead of a system connection.
This is almost wrong in all cases. In regular linux installation bluez will always be available on system bus, not on session bus.
You can see the error message in the exception exactly saying that there is no bluez service on your session bus.

abunaiquad, 2020-07-27:
Apologies on the formatting. 

When I do not include any filter, I still don't see my device. By looking at `btmon` log, I noticed that all the devices `bluez-dbus` found has one thing in common is 
> Flags: 0x06
 LE General Discoverable Mode

The device I am looking for only has 
> Flags: 0x04
          BR/EDR Not Supported

Any thoughts on what I can do to find my device?

Is there a javadoc on bluez-bus APIs?

Thanks much again

hypfvieh, 2020-07-27:
bluez-dbus only utilizes the regular bluez DBUS API, so almost anything found in the [bluez documentation](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc) is also valid for bluez-dbus.

Have you checked if you can find your device using the bluetoothctl commandline provided by bluez?

abunaiquad, 2020-07-27:
When I use `bluetoothctl` without modifying any setting as I access it, I don't see my device returning by the tool either.

However, when I change the following setting, then I can see my device

> menu scan -> discoverable on

Looking at `journalctl` log when I change setting (as state above), I see this:

> bluetoothd[16723]: src/adapter.c:parse_discovery_filter_dict() filtered discovery params: transport: 7 rssi: 32767 pathloss: 32767  duplicate data: false discoverable true

Running bluez-dbus codes I gave you earlier w/o setting filters, I get the following in `journalctl` 
bluetoothd[16723]: src/adapter.c:parse_discovery_filter_dict() filtered discovery params: transport: 6 rssi: 32767 pathloss: 32767  duplicate data: false discoverable false

Notice the differences between the filter that sees my device and the one that doesn't is `discoverable true`. Could you help explain what `discoverable true` mean? And is my assumption correct that setting `discoverable true` with your library then I will be able to see my device? If so, can you please show me how?

I should tell you now that from `bluetootctl`, I was able to connect to my device but after like 10 secs or so, bluetooth connection gets disconnected. 

Message from `bluetoothctl`
> ServicesResolved: no

Message from `btmon`
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                    #358 [hci0] 30597.196506
        Status: Success (0x00)
        Handle: 4
        Reason: Remote User Terminated Connection (0x13)

I don't know why this is happening. If you have any ideas or suggestions, I would be glad to know.

Sorry for going back to documentation question, I looked in bluez documentation that you posted above, I didn't see `DeviceManager.createInstance` information. Is that located somewhere?

Sorry for the long comment and as always thanks much again!

hypfvieh, 2020-07-27:
Switching discoverable to true does not make sense at all.
The discoverable flag just says if your adapter (your PC) will be discoverable by other bluetooth devices if they are scanning for devices (e.g. if you want to connect your smartphone to your PC, the PC will then be visible on your phone).

If you want to scan for devices in bluetoothctl, you just have to enter 'scan on'.

Regarding your question about DeviceManager: This is not part of bluez, it is a simple wrapper I wrote to add some sort of abstraction to bluez and get better java-like handling for bluez.
All documentation available is in that class.

abunaiquad, 2020-07-28:
For some reason, now I can see my device. I don't know what I did for it to do that. 

If you have any suggestion on why I am seeing `ServicesResolved: yes -> ServicesResolved: no`, please let me know.

I noticed that there is a couple new classes such as ProfileManager compared to v0.1.2, do you know when you might release it to mvn repository? Not sure if I need it for ServicesResolved issue.

Thanks much.

hypfvieh, 2020-07-28:
From the [bluez documentation](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt):

> bool ServicesResolved [readonly]
>
> Indicate whether or not service discovery has been
> resolved.

I don't have any release plans for the next few month. Maybe I'll do a release if the next version of dbus-java is ready to go.

abunaiquad, 2020-07-28:
Sorry for reopening this. There is a `discoverable filter` and I noticed that when that is `true`, `bluez-dbus` can see my device. I looked on where to set that in this library, and I couldn't find it. Is it missing? Otherwise, any help on how I can set `discoverable true` is greatly appreciated. Thanks.

Extract from `journalctl`:
bluetoothd[4517]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 1

Bluez codes on `current_discovery_filter`:
https://github.com/bluez/bluez/blob/a081a337bddd68924e3f48835c5e09f1bdbaeef9/src/adapter.c#L179
https://github.com/bluez/bluez/blob/a081a337bddd68924e3f48835c5e09f1bdbaeef9/src/adapter.c#L2409

Example of when adapter->current_discovery_filter == 1

> src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 1
bluetoothd[4517]: src/adapter.c:start_discovery_timeout() sending MGMT_OP_START_SERVICE_DISCOVERY 127, 7, 0
bluetoothd[4517]: src/adapter.c:start_discovery_complete() status 0x00
bluetoothd[4517]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1 method 1
bluetoothd[4517]: src/adapter.c:device_found_callback() hci0 addr 41:FC:C2:11:AA:3A, rssi -85 flags 0x0004 eir_len 28
bluetoothd[4517]: src/device.c:device_create() dst 41:FC:C2:11:AA:3A
bluetoothd[4517]: src/device.c:device_new() address 41:FC:C2:11:AA:3A

Example of when adapter->current_discovery_filter == 0
> src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 0
bluetoothd[4517]: src/adapter.c:start_discovery_complete() status 0x00
bluetoothd[4517]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1 method 0
bluetoothd[4517]: src/adapter.c:device_found_callback() hci0 addr 45:D7:D1:EE:02:E5, rssi -86 flags 0x0000 eir_len 17
bluetoothd[4517]: src/device.c:device_set_legacy() legacy 0
bluetoothd[4517]: src/device.c:device_set_rssi_with_delta() rssi -86
bluetoothd[4517]: src/device.c:device_set_tx_power() tx_power 12
bluetoothd[4517]: src/device.c:device_set_flags() flags 26
bluetoothd[4517]: src/adapter.c:device_found_callback() hci0 addr 6E:E9:A8:ED:18:0A, rssi -78 flags 0x0000 eir_len 14

I know they are different bluetooth devices. but notice `device_create()` and `device_new()` doesn't get called when current_discovery_filter = 0

hypfvieh, 2020-07-29:
`discoverable` is a property, [just set it](https://github.com/hypfvieh/bluez-dbus/blob/master/bluez-dbus/src/main/java/com/github/hypfvieh/bluetooth/wrapper/BluetoothAdapter.java#L125).

You should also start reading the source code. For me it feels like you never have looked at it.
You may also have a look at [my sandbox](https://github.com/hypfvieh/sandbox/tree/master/src/main/java/com/github/hypfvieh/sandbox/bluez).

closed 2020-07-29 by hypfvieh

abunaiquad, 2020-07-29:
Whatever is going on is very unstable. I did try to set `discoverable` to true but it still couldn't find my device.

> 1:25:49.769 INFO  (utils.TestListener-44) [TEST_METHOD_STARTED: bluez_dbus : []].  
11:25:50.121 INFO  (ble.Test-42) isDiscoverable false 
11:25:50.265 INFO  (ble.Test-44) isDiscoverable true 
11:25:50.266 INFO  (ble.Test-72) scanning... 
11:26:10.826 INFO  (ble.Test-75) [hci0] 65:4A:12:6D:BA:B4 65-4A-12-6D-BA-B4 
11:26:10.832 INFO  (ble.Test-75) [hci0] 6E:E9:A8:ED:18:0A 6E-E9-A8-ED-18-0A 
java.lang.IllegalStateException: device not found.
	at ble.Test.bluez_dbus(Test.kt:81)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.testng.internal.MethodInvocationHelper.invokeMethod(MethodInvocationHelper.java:124)
	at org.testng.internal.Invoker.invokeMethod(Invoker.java:580)
	at org.testng.internal.Invoker.invokeTestMethod(Invoker.java:716)
	at org.testng.internal.Invoker.invokeTestMethods(Invoker.java:988)
	at org.testng.internal.TestMethodWorker.invokeTestMethods(TestMethodWorker.java:125)
	at org.testng.internal.TestMethodWorker.run(TestMethodWorker.java:109)
	at org.testng.TestRunner.privateRun(TestRunner.java:648)
	at org.testng.TestRunner.run(TestRunner.java:505)
	at org.testng.SuiteRunner.runTest(SuiteRunner.java:455)
	at org.testng.SuiteRunner.runSequentially(SuiteRunner.java:450)
	at org.testng.SuiteRunner.privateRun(SuiteRunner.java:415)
	at org.testng.SuiteRunner.run(SuiteRunner.java:364)
	at org.testng.SuiteRunnerWorker.runSuite(SuiteRunnerWorker.java:52)
	at org.testng.SuiteRunnerWorker.run(SuiteRunnerWorker.java:84)
	at org.testng.TestNG.runSuitesSequentially(TestNG.java:1208)
	at org.testng.TestNG.runSuitesLocally(TestNG.java:1137)
	at org.testng.TestNG.runSuites(TestNG.java:1049)
	at org.testng.TestNG.run(TestNG.java:1017)
	at com.intellij.rt.testng.IDEARemoteTestNG.run(IDEARemoteTestNG.java:66)
	at com.intellij.rt.testng.RemoteTestNGStarter.main(RemoteTestNGStarter.java:110)

hypfvieh, 2020-07-29:
I still don't understand what you are trying to do or what kind of device you are hoping to find.

Setting discoverable flag on the bluetooth adapter does not change anything, like I said before.
This flag is used to make your adapter visible to other devices if the other device is scanning for devices.
It will not change any behaviour when your adapter is scanning for devices.