hypfvieh/bluez-dbus #40: How to scan the specified address device

sxyandapp, 2020-08-08:
I just  want   to  scan  the  specified  mac address device and  get the  service data.
I  tried to modify  the `DiscoveryFilter`  class  and  add  `Pattern(String.class)` , and  modify the `BluetoothAdapter`  and  add  `supportedFilterOptions.put("Pattern", String.class);`
but  i get  the below  error :
```
Exception in thread "main" org.freedesktop.dbus.exceptions.DBusExecutionException: Invalid arguments in method call
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at org.freedesktop.dbus.errors.Error.getException(Error.java:157)
        at org.freedesktop.dbus.errors.Error.throwException(Error.java:187)
        at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:164)
        at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:228)
        at com.sun.proxy.$Proxy19.SetDiscoveryFilter(Unknown Source)
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothAdapter.setDiscoveryFilter(BluetoothAdapter.java:364)
        at com.github.hypfvieh.bluetooth.DeviceManager.setScanFilter(DeviceManager.java:238)
```

is   the  dbus  not  supported  `Pattern`  or  other  reason ?

hypfvieh, 2020-08-08:
The 'Pattern' filter type was added to upstream bluez library 2020-02-19, so it will only work if you have a very recent version of bluez installed on your system (bluez 5.54, announced mid of march 2020).
Otherwise this operation will fail.

I would suggest not to use this scan filter if your application will be public, or you will get a lot of reports because the installed bluez version is too old.

sxyandapp, 2020-08-08:
> The 'Pattern' filter type was added to upstream bluez library 2020-02-19, so it will only work if you have a very recent version of bluez installed on your system (bluez 5.54, announced mid of march [...]

I just use it in  my  own  application, the version of bluez  on my device is 5.43,  thanks for you advice,  i  will  upgrade  the  bluez  version to verify it

closed 2020-08-08 by sxyandapp