hypfvieh/bluez-dbus #45: Scan Response

drofwarcs, 2020-09-11:
Hello,

Thanks for the wonderful library. I've been using it to setup a PI as a peripheral and its working great. 

I'm curious if your library (or Bluez in general) exposes any api's to add a scan response to advertisements. I would like to mimic an advertisement payload that exceeds the default 31 byte advertisement limits and the bluetooth spec allows this through a scan response. 

I found the following references a good intro to Scan Responses
https://www.novelbits.io/bluetooth-low-energy-advertisements-part-1/
https://www.novelbits.io/bluetooth-low-energy-advertisements-part-2/

hypfvieh, 2020-09-12:
Sorry I don't know. 
bluez-dbus is just a library which uses the interfaces published by bluez on DBus.

If bluez is capable of doing what you want to do, it should be possible using bluez-dbus as well as long as the required API is published on DBus.

drofwarcs, 2020-09-14:
Ok, thanks. Just found out that its still on their todo list https://github.com/bluez/bluez/blob/master/TODO .

closed 2020-09-14 by drofwarcs