hypfvieh/bluez-dbus #46: GATTService address wrong ?

sebpiller, 2020-09-16:
Very interesting library, by far the best I have found so far, in term of code quality, dependency management, and clarity...

A little question though, I have a working gatttool command like this, running on a Raspberry Pi 4:

    gatttool --device=C4:AC:05:42:73:A4 -t random --char-write-req -a 0x1f -n a001037F

... which set the brightness of a smart lamp to 100%. Note the "-a 0x1f", which corresponds to "char value handle" in gattool "characteristics"

    handle: 0x001e, char properties: 0x28, char value handle: **0x001f**, uuid: **44092842-0567-11e6-b862-0002a5d5c51b**

I try to make the same using bluez-dbus in java. My implementation seems correct, but with dbus-monitor I can see the following logs :

    method call time=1600276508.729104 sender=:1.184 -> destination=org.bluez serial=210 path=/org/bluez/hci0/dev_C4_AC_05_42_73_A4/service001d/**char001e**; interface=org.bluez.GattCharacteristic1; member=WriteValue
       array of bytes [
          0a 01 03 7f
       ]
       array [
       ]
    method return time=1600276508.776261 sender=:1.5 -> destination=:1.184 serial=6589 reply_serial=210

It looks I have rather to use the value of "char value handle", in my case 0x001f. 

Do you know if this is a bad usage of the library, an error on the device, or what ? 

Here is a little excerpt of the code, if you need more you can look here: https://github.com/sebpiller/luke-roberts-lamp-f

                BluetoothDevice lampF = manager.getDevices(true)
                    .stream()
                    .filter(e -> Objects.equals(e.getAddress(), config.getMac()))
                    .findFirst()
                    .get();
    ....
            String uuid = config.getCustomControlService().getUuid();
            BluetoothGattService customControlService = Objects.requireNonNull(lampF.getGattServiceByUuid(uuid));
            LOG.info("found GATT custom control service {} at UUID {}", customControlService, uuid);
    ....
            String externalApiUuid = config.getCustomControlService().getUserExternalApiEndpoint().getUuid();
            externalApi = Objects.requireNonNull(customControlService.getGattCharacteristicByUuid(externalApiUuid));

    ...
    private void sendCommandToExternalApi(LukeRoberts.LampF.Command command, Byte... parameters) {
        reconnectedIfNeeded();

        try {
            externalApi.writeValue(/*reversed*/ command.toByteArray(parameters), Collections.emptyMap());
        } catch (DBusException e) {
            throw new IllegalStateException("unable to change brightness: " + e, e);
        }
    }

Thanks for your time !

hypfvieh, 2020-09-17:
As far as I can see, you have to use 0x001f. The `gatttool` is doing some kind of padding, so whatever you pass to `-a` it is re-formatted to the format `0x0000` (see `gatttool --help-char-read-write`).

I don't have the time right now to lookup this in gatttool source, but I'm pretty sure that they do reformatting.

daviessm, 2020-09-17:
This sounds like problems I had when I started using BLE with BlueZ - some of the handles and UUIDs of characteristics didn't seem to match up.

Unfortunately it's so long ago I can't remember what I did to figure it all out, but I do remember using [DFeet](https://wiki.gnome.org/Apps/DFeet) to examine what DBus sees - if you can, I'd start there.

sebpiller, 2020-09-18:
Problem solved. It was a problem with my brain not able to see a typo: 0x0a != 0xa0. Sorry for the noise.

closed 2020-09-18 by sebpiller