hypfvieh/bluez-dbus #48: Ability to remove property and signal handlers from the com.github.hypfvieh.bluetooth.DeviceManager

OlegAndreych, 2020-10-01:
I have a case when the need for subscriptions for different properties comes and goes.

As a workaround, I can have a generic subscription, which will delegate to specific handlers, relieving one from the need to register handlers in the `DeviceManager`. But having only handlers with a lifecycle tied to the objects representing devices seems more natural.

As far as I can see, underlying org.freedesktop.dbus.connections.impl.DBusConnection already has corresponding methods.

hypfvieh, 2020-10-02:
I added methods to remove signal handlers.
But you can already do that without these methods by calling `deviceManager.getDBusConnection().removeSigHandler(class, handler)`

OlegAndreych, 2020-10-02:
> But you can already do that without these methods by calling deviceManager.getDBusConnection().removeSigHandler(class, handler)

Thanks for the methods and the idea.

closed 2020-10-02 by hypfvieh