hypfvieh/bluez-dbus #49: Can I write a descriptor to enable notification？

fanggw, 2020-10-19:
Now we use bluez-dbus on raspberry to communicate with smartlock. we have two types of smart locks.
Follow the same steps below, one type can receive the notification.but the other type cannot receive it all the time. 

<java code step>
1. deviceManager.registerPropertyHandler(new AbstractPropertiesChangedHandler() {
             // read notification
    }
2. startDiscovery  
3. bluetoothDevice.connect
4. bluetoothDevice.getGattServiceByUuid
5. notifyCharacteristic = bluetoothGattService.getGattCharacteristicByUuid("<notifyCharacteristic>");
    writeCharacteristic = bluetoothGattService.getGattCharacteristicByUuid("<writeCharacteristic>");
6. # notifyCharacteristic.getGattDescriptorByUuid("..").writeValue(new byte[] {0x01, 0x00}, null);     // handle 000c
    # writeValue reported error.   org.bluez.Error.NotPermitted
    notifyCharacteristic.startNotify()

However, there is no problem using gattool directly.
<gatttool>
gatttool -b E6:F9:65:B9:0B:98 -t random -I --listen
>connect
>char-desc 0001
      ...
      handle: 0x000c, uuid: 00002902-0000-1000-8000-00805f9b34fb
      ...
>char-write-req 000c 0100
## There is no NotPermitted error, the write is successful, and you can receive the notification immediately

How can I write a descriptor to enable notification? Or, in addition to startnotify, what else do I have to do to get the notification?

Thanks.

fanggw, 2020-10-20:
I don't know the reason, but after the above operation, I can normally receive the notification by executing disconnect / connect again.

hypfvieh, 2020-10-20:
Sorry I don't know whats going on. Usually you should be able to send the exact same commands/parameter to the characteristics like gatttool does. 
A more complete sample could be helpful as well.

bem22, 2021-02-11:
@fanggw to me it looks like you are using python. I managed to receive notifications in python (pygatt). Let me know if this is the case.

closed 2021-03-02 by hypfvieh