hypfvieh/bluez-dbus #51: Question Mesh API support

ddovnar, 2020-12-25:
Hi! Can anybody tell something about support Mesh API? How long it make or when it will be available? Thank's

hypfvieh, 2020-12-28:
I just added the missing interfaces. I'm not 100% sure if they are correct. 
It also seems that the API is not really final right now, as it is not part of bluez 5.54 and is only available in source repos...

ddovnar, 2021-02-20:
Hi! Thanks for quick answer. Now I can test your functionality by working with mesh network, but I don't understand how can I use it. In my case, I need bind AppKey and NetworkKey, and then connect to mesh network. Next step, I must receive data from sensors by SensorClient model. Do you have, any suggest or example of code, to do that?

ddovnar, 2021-02-20:
I found similar issue in python [python-bluetooth-mesh](https://github.com/SilvairGit/python-bluetooth-mesh).

hypfvieh, 2021-02-21:
Sorry can't help you with this. I don't use mesh and I never spend time understanding how it works and where it can be used.
This is one of two reasons why mesh was not part of bluez-dbus yet. The other reason was that it is not included in any stable release of bluez.

closed 2023-02-20 by hypfvieh