hypfvieh/bluez-dbus #57: How to set bluetooth mtu

1784450277, 2023-03-14:
I want to set the mtu of Bluetooth and call the gattCharacteristic.AcquireWrite method, but an error is reported
```
code：
  Map<String, Variant<?>> ops = new HashMap<>();
  ops.put("mtu", new Variant<>(240));
  TwoTuple<FileDescriptor, UInt16> twoTuple = writeService.getRawGattCharacteristic().AcquireWrite(ops);

error:
java.lang.NullPointerException
	at org.freedesktop.dbus.messages.Message.extractOne(Message.java:982)
	at org.freedesktop.dbus.messages.Message.extract(Message.java:1145)
	at org.freedesktop.dbus.messages.Message.extract(Message.java:1122)
	at org.freedesktop.dbus.messages.Message.getParameters(Message.java:1253)
	at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:156)
	at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:216)
	at com.sun.proxy.$Proxy23.AcquireWrite(Unknown Source)
	at org.ScanMain.test(ScanMain.java:112)
	at org.ScanMain.main(ScanMain.java:41)

hypfvieh, 2023-03-14:
If you want to work with file descriptors you have to add another dependency to your project: `com.rm5248:dbus-java-nativefd:1.0`
DBus-java does not support file descriptors out of the box.

1784450277, 2023-03-14:
I'm sorry to trouble you again. What I want to do is set the mtu value of Bluetooth. When using gattCharacteristic.AcquireWrite, I also see the relevant mtu introduction in the document, so I try. I don't know if I need to use file descriptors. Can you help me with setting mtu? I haven't seen the relevant introduction on the official website of Bluez

hypfvieh, 2023-03-14:
You use file descriptors because the call to `AcquireWrite` returns a Tuple containing a file descriptor. This can only be created/serialized when file descriptor support is enabled.

I'm also not sure that `AcquireWrite` is really what you want to do. Usually I would try using `WriteValue` instead.
Also note that changing the MTU is not always possible. Some device do not allow that, in most cases it is only possible if "you are the server", which means you provide the GATT characteristics.

closed 2023-03-27 by 1784450277