hypfvieh/bluez-dbus #58: Unable to get Metadata PropertiesChanged of org.mpris.MediaPlayer2.Player

moontide, 2023-03-17:
I try to get Metadata PropertiesChanged data of firefox media player, I can see it's information in command output of `dbus-monitor`.

    signal time=1679021922.354447 sender=:1.81 -> destination=(null destination) serial=605 path=/org/mpris/MediaPlayer2; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
       string "org.mpris.MediaPlayer2.Player"
       array [
          dict entry(
             string "Metadata"
             variant             array [
                   dict entry(
                      string "mpris:trackid"
                      variant                      object path "/org/mpris/MediaPlayer2/firefox"
                   )
                   dict entry(
                      string "xesam:title"
                      variant                      string "Romance"
                   )
                   dict entry(
                      string "xesam:album"
                      variant                      string "La Voix Des Anges"
                   )
                   dict entry(
                      string "xesam:artist"
                      variant                      array [
                            string "Dominica"
                         ]
                   )
                ]
          )
       ]
       array [
       ]

    signal time=1679039076.133555 sender=:1.81 -> destination=(null destination) serial=720 path=/org/mpris/MediaPlayer2; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
       string "org.mpris.MediaPlayer2.Player"
       array [
          dict entry(
             string "PlaybackStatus"
             variant             string "Paused"
          )
       ]
       array [
       ]

But I can't get this `Metadata` information in dbus-java.

I added `System.out.println (_s.getPropertiesChanged());` [here in SimpleBluez.java](https://github.com/hypfvieh/sandbox/blob/master/src/main/java/com/github/hypfvieh/sandbox/bluez/SimpleBluez.java#L132 ), but I only see the `PlaybackStatus` information in the output when playing music in firefox, no `Metadata` information.

### Environments
* OS: Fedora 37 x86_64
* dbus-java: 3.3.2
* bluez-dbus: 0.1.4
*  *(The following packages are all installed from fedora official repository)*
* bluez: 5.66
* Firefox: 110.0
* dbus: 1.14.6

hypfvieh, 2023-03-17:
First: this is not related to bluez-dbus because the mpris stuff has nothing to do with bluetooth or bluez.

I assume you are not listening to the correct dbus bus.
Bluez is usually found on system bus, while most of the things happening in user context is on session bus.
The sample you referenced uses the system bus only to catch properties changed events issued by/for bluez related actions.
As I don't have the mpris stuff on my Ubuntu machine, I cannot check, but I would check if the desired information is found on session bus.

moontide, 2023-03-17:
> First: this is not related to bluez-dbus because the mpris stuff has nothing to do with bluetooth or bluez.
> 
> I assume you are not listening to the correct dbus bus. Bluez is usually found on system bus, while most of the things happening in user context is on session bus. 

Sorry, I'm pretty new to dbus, didn't know there's system bus and there's session bus.

> The sample you referenced uses the system bus only to catch properties changed events issued by/for bluez related actions.

I added the code before filter out code, I can see information about `org.freedesktop.NetworkManager` or `org.freedesktop.systemd1.Service` been printed, but no `Metadata` information of `org.mpris.MediaPlayer2.Player`. So I guess your assumption could be right. But I do see `PlaybackStatus` information of `org.mpris.MediaPlayer2.Player`, could firefox send these two information to different bus? 

> As I don't have the mpris stuff on my Ubuntu machine, I cannot check, but I would check if the desired information is found on session bus.

The [mpris interface](https://specifications.freedesktop.org/mpris-spec/2.2/index.html) stuff seems [been implemented in firefox 75 already](https://bugzilla.mozilla.org/show_bug.cgi?id=1353652). I also tested Audacious media player, I also see information about `org.mpris.MediaPlayer2.Player` in `dbus-monitor` command ouput.

hypfvieh, 2023-03-17:
Firefox does not provide mpris on my machine, nothing to see in d-feet on system or session bus.
When starting audacious, I can see the mpris stuff in the audacious process on session bus `org.mpris.MediaPlayer2.audacious`.
So it looks like you have to add a listener to the session bus.

moontide, 2023-03-18:
> Firefox does not provide mpris on my machine, nothing to see in d-feet on system or session bus.

I guess you need visit a URL contains media content (say, youtube), and play it.

> So it looks like you have to add a listener to the session bus.

Yes, you're totally correct. I test it today, and I got the `Metadata` information as expected from session bus.

closed 2023-03-18 by moontide