hypfvieh/bluez-dbus #59: get RSSI fail

1784450277, 2023-04-03:
bluetoothDevice.getRssi()
Method threw 'org.freedesktop.dbus.exceptions.DBusExecutionException' exception.

[image]

hypfvieh, 2023-04-03:
So what do you trying to tell me?
The exception clearly says, there is no RSSI property on the device you want to query. 
Not every device supports every feature or property, so the best thing the libary can do is throwing an exception when something is not supported.

1784450277, 2023-04-04:
Sorry, I originally thought this attribute was present on all devices because the corresponding RSSI value can be obtained during scanning. Thank you for your reply

closed 2023-04-04 by 1784450277