hypfvieh/bluez-dbus #60: Bluetooth scanning can scan for devices that are turned off

1784450277, 2023-04-18:
I found that when Bluetooth scans, it will scan devices that have already been turned off. Have you ever encountered this problem before? This device was previously connected by me

com.github.hypfvieh.bluetooth.DeviceManager#scanForBluetoothDevices(java.lang.String, int)

hypfvieh, 2023-04-18:
Which devices are visible/found during scan depends on what bluez sees. 
When bluez thinks that a device is still available (even though it has been turned off) it will be found during scan.

hypfvieh, 2023-04-18:
This library is only a bridge between java and bluez. It does this by providing the required interface classes used to connect to bluez using dbus.
If bluez has stored something somewhere and still propagates this information on dbus you will still get that information.

There is no way of clearing this data. This library simply uses what bluez is publishing on DBus. You may try the bluez commandline tool to check what bluez does. You may also use dbus commandline utils (or a GUI like d-feet) to see what information bluez is publishing on the bus.

1784450277, 2023-04-20:
I have found a solution that can clear the cache after executing a command
    service bluetooth stop
    rm -rf /var/lib/bluetooth
    service bluetooth start

closed 2023-04-20 by 1784450277