hypfvieh/bluez-dbus #61: Method "Disconnect" on interface "org.bluez.Device1" doesn't exist

lsiepel, 2023-09-06:
The bluetooth.bluez binding that is used by openHAB depends on this library. Some reports are made of an exception with this stacktrace:

```
2023-09-05 18:14:28.332 [ERROR] [core.thing.internal.ThingManagerImpl] - Exception occurred while disposing handler of thing 'bluetooth:ruuvitag_beacon:hci0:e12cd47b86a8': Method "Disconnect" with signature "" on interface "org.bluez.Device1" doesn't exist
org.freedesktop.dbus.errors.UnknownObject: Method "Disconnect" with signature "" on interface "org.bluez.Device1" doesn't exist
        at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77) ~[?:?]
        at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
        at java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499) ~[?:?]
        at java.lang.reflect.Constructor.newInstance(Constructor.java:480) ~[?:?]
        at org.freedesktop.dbus.errors.Error.getException(Error.java:157) ~[?:?]
        at org.freedesktop.dbus.errors.Error.throwException(Error.java:187) ~[?:?]
        at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:164) ~[?:?]
        at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:228) ~[?:?]
        at jdk.proxy32.$Proxy194.Disconnect(Unknown Source) ~[?:?]
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.disconnect(BluetoothDevice.java:434) ~[?:?]
        at org.openhab.binding.bluetooth.bluez.internal.BlueZBluetoothDevice.disconnect(BlueZBluetoothDevice.java:199) ~[?:?]
        at org.openhab.binding.bluetooth.BeaconBluetoothHandler.dispose(BeaconBluetoothHandler.java:139) ~[?:?]
        at org.openhab.binding.bluetooth.ruuvitag.internal.RuuviTagHandler.dispose(RuuviTagHandler.java:88) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:568) ~[?:?]
        at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:147) ~[?:?]
        at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) ~[?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[?:?]
        at java.lang.Thread.run(Thread.java:833) ~[?:?]
```

It seems that the library assumes the device actually has a `Disconnect` method while it obviously doesn't. See here: https://github.com/hypfvieh/bluez-dbus/blob/0e49c67d8578984cad0de174faacdcdbc435bce7/bluez-dbus/src/main/java/com/github/hypfvieh/bluetooth/wrapper/BluetoothDevice.java#L434C36-L434C36

I double checked and it looks like the binding retrieves all devices from this libs `DeviceManager.getDevices` So i'm not sure where in this loop a device is introduced that does not (properly) implement the `Disconnect` method

hypfvieh, 2023-09-07:
Difficult to say without having any of those problematic devices here for testing.

The library just queries DBus which in turn is connected to Linux bluez stack. 

So I would suggest starting with DBus and bluez.
First, please check if you can see the problematic device on DBus (which should be the case because otherwise bluez-dbus wouldn't see that as well).
When you have found the device, try to call the disconnect method directly on DBus (using d-feet, dbus-send or other DBus utils).
If that does not give you any error, than it must be a problem with bluez-dbus.
Otherwise it is a problem of bluez, or of the device and you pretty much out of luck to fix it in bluez-dbus.

pbloigu, 2024-01-01:
> Difficult to say without having any of those problematic devices here for testing.
>
> The library just queries DBus which in turn is connected to Linux bluez stack.
>
> So I would suggest starting with [...]

I experienced this issue with [RuuviTags](https://ruuvi.com/ruuvitag/) in the context of Openhab. I could be able to debug this further but I'm not at all familiar with those DBus tools mentioned so I would need some hand-holding in the process.

However, these devices are simple Bluetooth LE beacons that just emit data to whoever is listening, so for me it's intuitively difficult to understand the semantics of "being connected" in this case. Then again, my knowledge of Bluetooth protocols is extremely limited.

Anyway, if you want me to grab some logs or run some commands to debug this, just let me know.

pbloigu, 2024-01-02:
Hi @hypfvieh and thank you so much for taking the time to explain this! I appreciate it a lot!

Now, I was able to actually run the Disconnect command on one of those RuuviTags. It's actually the same device for which the stack trace is shown in the issue description. Here's a screenshot of what happened:

[image]
 
I'm not sure what should happen, but at least there's no clear error visible in there. I don't know what to make of this. 

However, this was on my Ubuntu laptop, while my Openhab instance (from which the stack trace is) is running on RPi4 with Raspbian. Should I check dbus / BlueZ versions on these machines if there's some difference there?

hypfvieh, 2024-01-02:
That is quite interesting. I would have expected to see some kind of error in the "Method output" box.
I've seen similar behaviors when using some BLE controlled LED strips. In that case the BLE connection of the devices were not very reliable, so sometimes when trying to call a method on the device it fails at first but worked when tried two or three times. 
I never got any real nice solution for this other than retrying and ignoring certain failures.

I'm also not sure if the bluez version will make any substantial difference. It might be interesting to see which versions were used on Ubuntu and on Raspi, but as long as it some 5.3x or higher it should make no real difference in this case.
  
Anyways, you may try the following on your Raspi (I guess that you don't have a graphical system but SSH access):

`dbus-send --print-reply --system --dest=org.bluez /org/bluez/hci0/dev_E1_2C_D4_7B_86_A8 org.bluez.Device1.Disconnect`

This should do basically the same as you did with d-feet. It will try to call the `Disconnect()` on the device (I used the path seen in the screenshot).
If something went wrong, you will see something like this:
`Error org.freedesktop.DBus.Error.UnknownObject: Method "Disconnect" with signature "" on interface "org.bluez.Device1" doesn't exist`
This is what happens when I try to execute that call - obviously because I don't have any device with the used path on my DBus instance. 
Would be interesting to know what happens on the Raspi.

pbloigu, 2024-01-03:
Hi!

I actually ssh'd to the RPi with X-Forwarding and ran d-feet over that. However, at first there were no devices listed in the path /org/bluez/hci0/dev_*. I had to run `bluetoothctl scan on` and then refresh the d-feet device list to see all Bluetooth devices within range. Only then I was able to successfully execute the Disconnect command on this particular device.

Then I stopped the Bluetooth scan with Ctrl+C and tried to execute Disconnect again on the same device and was greeted with the "method doesn't exist" error. Which kinda makes sense since after refreshing the device list in d-feet again, all Bluetooth devices were again gone.

I executed all the steps above with Openhab service stopped. However, when I started the Openhab instance and refreshed the device list in d-feet, they were all there again so I'm guessing Openhab initiates the scan by itself, but then at some point stops it for some reason resulting in Bluetooth devices becoming unavailable over DBus. Also, stopping the Openhab instance makes all devices disappear from d-feet (after refreshing the device list). 

Therefore I think we can conclude with a fair confidence that this issue is not with bluez-dbus library.

hypfvieh, 2024-01-04:
This completely makes sense. 

bluez-dbus can only work with the devices seen on the bus. If a device disappears by any reason and bluez does not recognize this immediately (or waits for some time to ensure the device is really gone), then this is the time frame were you may try to execute any action on the device and will receive errors.
Also when using some caching behavior (look for devices once, then remember what was seen using the results later) may also cause this problem. 

Thanks for your support @pbloigu - you did exactly what I would have done. Using d-feet or similar to directly "talk" to DBus/bluez will eliminate all possible problems which may be related to or introduced by dbus-java and bluez-dbus. 

I would say everything is working "as designed".

@lsiepel do you agree? Can this issue be closed?

lsiepel, 2024-01-04:
Yes, let's close it here and we will try to continue our work in the openHAB repo. Just as a note we did introduce a functional workaround, but that only hides the underlying problem. Thanks for your support!

closed 2024-01-04 by lsiepel