hypfvieh/bluez-dbus #62: bluez-dbus-osgi 2.0.0

lsiepel, 2024-07-21:
Thanks again for the explenation you provided here:
https://github.com/openhab/openhab-addons/issues/17035#issuecomment-2241152882

I have now looked into this further and concluded that there was some package mixup. openHAB uses `bluez-dbus-osgi` version 0.1.4 
The readme in this repo showed that a 2.0.0 version was in development supporting the more recent 4.x brnach of dbus-java.

Is there any change these packages get released?

hypfvieh, 2024-07-21:
Mhh... I guess bluez-dbus did not the attention it should. I simply forgot to create any release :facepalm: 

Anyway, I have released 0.2.0 now which will use dbus-java 4.3.x  (requires Java 11). 
The release should be available until tomorrow (depending on how fast Maven Central syncs there servers). 

I'm planning to release 0.3.0 of bluez-dbus when dbus-java 5.1.0 is released which may be done in a few weeks. The 0.3.x will then require Java 17.

lsiepel, 2024-07-21:
Ah, too bad. Thanks for the release! If i understand it right, the 0.2.0 (with 4.x dependency)  is a drop in replacement, but 0.3.0 (with 5.1 dependency) is not.
Is there some upgrade path / documentation about breaking changes i can use?

openHAB uses 1java 17 as minimum, so i don't see issues there.

hypfvieh, 2024-07-21:
If it is a drop-in replacement depends on how everything is used.
All interfaces provided by this library are still the same. The dependency on dbus-java is needed for the DBusInterface etc. 

The structure of dbus-java changed between 3.x and 4.x. 
While 3.x was much like the original dbus-java version 2.7 but without libmatthew, 4.x introduced the usage of "transports". 
A transport defines an implementation of some protocol to communicate with the DBus Daemon.
DBus usually uses Unixsockets to communicate, but it also is capable of using TCP (but that is disabled in default DBus configuration). 
I know that some user of dbus-java also implemented a transport which uses SSH.

In the beginning of major version 4 of dbus-java there were 2 transports, jnr-based Unixsockets and TCP.
Some time later the transport based on native UnixSocket of the JDK was added when Java 16 arrived.
And then, again some time later someone provided a transport implementation for Unixsockets using junixsocket library.

In major version 5, I put some effort in refactoring, removing deprecated stuff, cleaning up the code and fixing a lot of bugs. In the latest version one could consider the implementation of junixsocket based transport as stable.  

So if you do not import any of the dbus-java classes/package directly and using the OSGi bundle, it should be like drop-in replacement. 
If you don't use OSGi you have to add a transport.

This is the same for the upcoming next version (0.3.x) based on dbus-java 5.x. 

The difference will be that I plan to change the default included transport for OSGi bundle from jnr-unixsockets to junitsocket implementation, so that the OSGi bundle should be able to support filedescriptor passing on DBus out of the box.

lsiepel, 2024-07-22:
Ok, i created a PR for updating openHAB to 0.2.0. All seems ok and a drop-in replacement for how we use it.
When the 0.3.0 version is available i might look int updating again, but without Bluez hardware it would be difficult to test and check the changes. Thanks for now!