hypfvieh/bluez-dbus #63: Example code for a GattServer

mtexplorative, 2024-08-15:
Hi,
we are trying to implement Bluetooth as an additional path to control an IoT-Device.
Our current set-up regarding the compute hardware is a Raspberry Pi (may changed later) but will still be Linux ;)
Our functionality is already accessible via REST-calls provided by a micronaut-based Java program.
I struggle in setting-up a GanttServer with this library.
I would appreciate it if a simple example code could be provided showing how to provide a read function on a property (incl. advertsing it).
Regards, Markus.

hypfvieh, 2024-08-15:
Providing a Gatt-Server is not really straight forward using bluez and dbus.
Without any warranty, here is some sample code, maybe you can start with that. Other samples in Python and C can be found in the [bluez.git](https://github.com/bluez/bluez/) Repository (see tools and test directory).

```java
package com.github.hypfvieh.sandbox.bluez;

import org.bluez.GattCharacteristic1;
import org.bluez.GattDescriptor1;
import org.bluez.GattManager1;
import org.bluez.GattService1;
import org.bluez.datatypes.TwoTuple;
import org.bluez.exceptions.*;
import org.freedesktop.dbus.DBusPath;
import org.freedesktop.dbus.FileDescriptor;
import org.freedesktop.dbus.connections.impl.DBusConnection;
import org.freedesktop.dbus.connections.impl.DBusConnectionBuilder;
import org.freedesktop.dbus.interfaces.DBusInterface;
import org.freedesktop.dbus.interfaces.Properties;
import org.freedesktop.dbus.types.UInt16;
import org.freedesktop.dbus.types.Variant;

import java.util.*;

public class GattExample {
    static final String FLAG_READ = "read";
    static final String FLAG_WRITE = "write";

    public static void main(String[] args) {

        MyGattApplication myGattApplication = new MyGattApplication();
        SampleService sampleService = new SampleService();
        SampleCharacteristics sampleCharacteristics = new SampleCharacteristics(sampleService);
        SampleGattDescriptor sampleDescriptor = new SampleGattDescriptor(sampleCharacteristics);
        sampleCharacteristics.getDescriptors().add(sampleDescriptor);
        myGattApplication.addService(sampleService);

        sampleService.getCharacteristics().add(sampleCharacteristics);

        try (DBusConnection dbus = DBusConnectionBuilder.forSystemBus().build()) {

            dbus.exportObject(myGattApplication);

            GattManager1 gattMgr = dbus.getRemoteObject("org.bluez", "/org/bluez/hci0", GattManager1.class);

            Map<String, Variant<?>> opts = new HashMap<>();

            gattMgr.RegisterApplication(new DBusPath(myGattApplication.getObjectPath()), opts);
            Thread.sleep(50000000L); // do something useful or a proper wait loop
        } catch (Exception _ex) {
            // TODO Auto-generated catch block
            _ex.printStackTrace();
        }
    }

    public static class MyGattApplication implements GattServiceApp {

        private final List<SampleService> services = new ArrayList<>();

        @Override
        public String getObjectPath() {
            return "/";
        }

        public void addService(SampleService _service) {
            services.add(_service);
        }

        @Override
        public Map<DBusPath, Map<String, Map<String, Variant<?>>>> GetManagedObjects() {

[... 264 lines cut ...]
                "Characteristic", new Variant<>(new DBusPath(characteristic.getObjectPath())),
                "Flags", new Variant<>(flags, "as"));
        }

    }

    public interface GattServiceApp extends DBusInterface {

        Map<DBusPath, Map<String, Map<String, Variant<?>>>> GetManagedObjects();
    }

    public interface PropertiesHelper extends Properties {

        Class<? extends DBusInterface> getBluezInterface();

        default Map<String, Map<String, Variant<?>>> getProps() {
            Map<String, Map<String, Variant<?>>> result = new LinkedHashMap<>();
            result.put(getBluezInterface().getName(), GetAll(null));
            return result;
        }

    }
}

```

mtexplorative, 2024-08-16:
Wow, thank you so much for providing this sketch. I guess I will need some time to work through it. This is exactly what I was looking for.
Regards, Markus.