hypfvieh/bluez-dbus #64: OSGi Bundle issues

lsiepel, 2025-01-19:
We seem to have problems with release 0.3.0 the OSGI packes seems to have issues we didn't experiecne that with 0.2.0
Are you aware if this? I also noticed something strange, so it might be that 0.1.4 was the last one that seem to work.

For reference:
https://github.com/openhab/openhab-addons/pull/18124

```
2025-01-19 19:05:51.638 [ERROR] [Events.Framework                    ] - FrameworkEvent ERROR
java.util.zip.ZipException: Exception in opening zip file: /var/lib/openhab/cache/org.eclipse.osgi/247/0/.cp/junixsocket-core-2.10.0.pom
        at org.eclipse.osgi.framework.util.SecureAction.getZipFile(SecureAction.java:356) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.storage.bundlefile.ZipBundleFile.doOpen(ZipBundleFile.java:51) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.storage.bundlefile.CloseableBundleFile.internalOpen(CloseableBundleFile.java:140) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.storage.bundlefile.CloseableBundleFile.lockOpen(CloseableBundleFile.java:78) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.storage.bundlefile.CloseableBundleFile.getEntry(CloseableBundleFile.java:274) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.storage.bundlefile.BundleFile.getResourceURL(BundleFile.java:142) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.classpath.ClasspathEntry.findResource(ClasspathEntry.java:213) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalResources(ClasspathManager.java:466) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalResources(ClasspathManager.java:452) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalResources(ModuleClassLoader.java:331) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.BundleLoader.findLocalResources(BundleLoader.java:922) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.BundleLoader.findResources(BundleLoader.java:781) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.loader.ModuleClassLoader.getResources(ModuleClassLoader.java:240) ~[org.eclipse.osgi-3.18.0.jar:?]
        at org.eclipse.osgi.internal.framework.ContextFinder.getResources(ContextFinder.java:182) ~[org.eclipse.osgi-3.18.0.jar:?]
        at java.util.ServiceLoader$LazyClassPathLookupIterator.nextProviderClass(ServiceLoader.java:1203) ~[?:?]
        at java.util.ServiceLoader$LazyClassPathLookupIterator.hasNextService(ServiceLoader.java:1228) ~[?:?]
        at java.util.ServiceLoader$LazyClassPathLookupIterator.hasNext(ServiceLoader.java:1273) ~[?:?]
        at java.util.ServiceLoader$2.hasNext(ServiceLoader.java:1309) ~[?:?]
        at java.util.ServiceLoader$3.hasNext(ServiceLoader.java:1393) ~[?:?]
        at javax.xml.xpath.$XPathFactoryFinder.lambda$findServiceProvider$18($XPathFactoryFinder.java:275) ~[?:?]
        at java.security.AccessController.doPrivileged(AccessController.java:319) ~[?:?]
        at javax.xml.xpath.$XPathFactoryFinder.findServiceProvider($XPathFactoryFinder.java:272) ~[?:?]
        at javax.xml.xpath.$XPathFactoryFinder._newFactory($XPathFactoryFinder.java:149) ~[?:?]
        at javax.xml.xpath.$XPathFactoryFinder.newFactory($XPathFactoryFinder.java:84) ~[?:?]
        at javax.xml.xpath.XPathFactory.newInstance(XPathFactory.java:69) ~[?:?]
        at javax.xml.xpath.XPathFactory.newInstance(XPathFactory.java:45) ~[?:?]
        at com.github.hypfvieh.XmlHelper.applyXpathExpressionToDocument(XmlHelper.java:67) ~[?:?]
        at com.github.hypfvieh.DbusHelper.findNodes(DbusHelper.java:43) ~[?:?]
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.refreshGattServices(BluetoothDevice.java:62) ~[?:?]
        at com.github.hypfvieh.bluetooth.wrapper.BluetoothDevice.getGattServices(BluetoothDevice.java:50) ~[?:?]
        at org.openhab.binding.bluetooth.bluez.internal.BlueZBluetoothDevice.discoverServices(BlueZBluetoothDevice.java:414) ~[?:?]
        at org.openhab.binding.bluetooth.bluez.internal.BlueZBluetoothDevice.updateBlueZDevice(BlueZBluetoothDevice.java:116) ~[?:?]
        at org.openhab.binding.bluetooth.bluez.internal.BlueZBridgeHandler.initializeAndRefreshDevices(BlueZBridgeHandler.java:194) ~[?:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:572) ~[?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:358) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642) ~[?:?]
        at java.lang.Thread.run(Thread.java:1583) ~[?:?]
Caused by: java.util.zip.ZipException: zip END header not found
        at java.util.zip.ZipFile$Source.findEND(ZipFile.java:1649) ~[?:?]
        at java.util.zip.ZipFile$Source.initCEN(ZipFile.java:1657) ~[?:?]
        at java.util.zip.ZipFile$Source.<init>(ZipFile.java:1495) ~[?:?]
        at java.util.zip.ZipFile$Source.get(ZipFile.java:1458) ~[?:?]
        at java.util.zip.ZipFile$CleanableResource.<init>(ZipFile.java:724) ~[?:?]
        at java.util.zip.ZipFile.<init>(ZipFile.java:251) ~[?:?]
        at java.util.zip.ZipFile.<init>(ZipFile.java:180) ~[?:?]
        at java.util.zip.ZipFile.<init>(ZipFile.java:194) ~[?:?]
        at org.eclipse.osgi.framework.util.SecureAction.getZipFile(SecureAction.java:342) ~[org.eclipse.osgi-3.18.0.jar:?]
        ... 38 more
```

hypfvieh, 2025-01-20:
No I'm not aware of any problems. 
I have to admit that I don't have any way to test the OSGi bundle nor have any clue how it works.
The OSGi packaging stuff was provided as PR by someone else years ago (same goes for OSGi bundle of dbus-java). 

I just take care my code is working and updating the dependencies/plugins from time to time.

There is no difference in packaging between 0.2.x and 0.3.x besides the usage of dbus-java 5.x instead of 4.x.

Between 0.1.x and 0.2.x the bundle-plugin was updated from 5.1.1 to 5.1.9 and also `jnr-unixsocket` and `tcp-transport` were added as OSGi dependency because otherwise the bundle didn't work (you need at least one transport).
Afaik that version was then successfully used in openHAB. 

Anyway, as said before, I'm not familiar with OSGi. If something is broken, help would be much appreciated.
If necessary I can create new releases for both dbus-java and bluez-dbus quickly.

hypfvieh, 2025-01-20:
@joerg1985 is this issue now fixed with your PR, should I create a new release?

joerg1985, 2025-01-20:
@hypfvieh looks good now, i think you can create a release.
Thanks for the fast response!

hypfvieh, 2025-01-20:
thanks. Release is done and should be available in Maven Central soon (usually it should be available tomorrow).
Please close this issue if it is solved with that patch + release.

joerg1985, 2025-01-21:
Looks like the fix did work, but i can not close the issue. @lsiepel could you please close it?

lsiepel, 2025-01-21:
Thanks all!

closed 2025-01-21 by lsiepel