hypfvieh/bluez-dbus #66: Error building pkg on Debian aarch64

qzic, 2025-01-23:
I'm trying to install this on a Debian Linux aarch64. I followed the instructions down to the build stage and get this error:
```
tools/rctest.c: In function ‘recv_mode’:
tools/rctest.c:507:47: error: ‘SIOCGSTAMP’ undeclared (first use in this function); did you mean ‘SIOCGARP’?
  507 |                                 if (ioctl(sk, SIOCGSTAMP, &tv) < 0) {
      |                                               ^~~~~~~~~~
      |                                               SIOCGARP
tools/rctest.c:507:47: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [Makefile:5842: tools/rctest.o] Error 1
make: *** [Makefile:9273: install] Error 2

****  Installation failed. Aborting package creation.
```
I'm no linux expert :-) Is there an easy fix?

I'm trying to replace Bluecove which I have running on a 32 bit machine but have given up 64bit.

qzic, 2025-01-23:
I solved the above undeclared symbol problem by adding #include "/usr/include/linux/sockios.h" to a few .c files. Now I have symbol conflicts eg bool pause(void *) between your code and /usr/include/unistd.h

```
  CC       obexd/client/obexd-sync.o
obexd/client/sync.c:225:26: error: ‘sync’ redeclared as different kind of symbol
  225 | static struct obc_driver sync = {
      |                          ^~~~
In file included from /usr/include/aarch64-linux-gnu/bits/sigstksz.h:24,
                 from /usr/include/signal.h:328,
                 from /usr/include/glib-2.0/glib/gbacktrace.h:38,
                 from /usr/include/glib-2.0/glib.h:36,
                 from obexd/client/sync.c:32:
/usr/include/unistd.h:1005:13: note: previous declaration of ‘sync’ with type ‘void(void)’
 1005 | extern void sync (void) __THROW;
      |             ^~~~

```
Starting to feel like I'm going down a rat hole.

hypfvieh, 2025-01-23:
What are you trying to do?

If you use Debian, just install bluez using apt-get if it isn't installed already.

If you want to build bluez yourself, you are on your own.
The Bluez library is not related to this repository. 

The library provided in this repo (bluez-dbus) is for Java and contains no C Code which needs compilation. It utilizes dbus and bluez to provide access to the Bluetooth stack in a Java application.

qzic, 2025-01-23:
I use bluetooth (BlueCove 2.1.1) to communicate between a java program running on a Debian Raspberry Pi (32bit OS) and an Android App. I would like to move the RPi java program to a Debian Raspberry with a 64bit OS (aarch64). I did find a .deb pkg for 64 bit Bluecove but I have not been able to get it working despite ensuring that all the same (but64bit) libs are running on the Raspberry. So ...... It seems that the Dbus-Bluez alternative is newer/better/supported so I'm trying to change from Bluecove to DBus-Bluez. Of course then I need to use a different API which is fine. Any help would be appreciated with this.

hypfvieh, 2025-01-24:
Ok... Didn't know BlueCove but it looks very ancient (at least what I've found on google)...

Anyway. Why do you want to compile bluez manually? 
As far as I can see, bluez is available as arm64 build via apt (-> http://ftp.us.debian.org/debian/pool/main/b/bluez/bluez_5.66-1+deb12u2_arm64.deb).

All you need is bluez installed (and dbus, but that is usually present on any Linux distribution of the last 15 years or will be pulled in when using apt). 
No need for any header files, C compiler etc.

qzic, 2025-01-24:
OK, so I'm sorry I caused this noise. I should not have raised this issue in the first place (don't need to build bluez) and will delete it if I can. 

I just need to educate myself on how dbus and bluez-dbus and how to use it to set up a bluetooth server on my Raspberry Pi.

closed 2025-01-24 by qzic