hypfvieh/bluez-dbus #69: Not receiving all the fragments in WriteHandler

PavanNittala, 2025-04-11:
I am hosting a BLE gatt server on my Peripheral and its in advertising mode .  The client which is a Wallet sends data on a particular characteristic which has Write-Without-Response properties set

MTU is negotiated at 517. I am receiving just the last chunk of data which is weird,  when i did hcidump , i see data in all fragments , btmon shows Data in just the last chunk. I am using a dbus-java and bluez-dbus  ports to validate all my testing using Java 1.8 VM .

Attached hcidump, btmon logs and wireshark dump . In this scenario, i received just the last chunk as shown in btmon which is

00e79f133f5135d687e3c550a6bd930e69362c4eb12e3ad3019418d2855fcd790394bff363f27e968edaae5319275169d4b166e5f9d06d927ad3b621085b49079181301df47c929a85674fda06a684a0f056b23d9f9c4392721ce5f71da716f8c3a1c39fc6bc6588ae8226f85e2d06e15eae9b126039f05cbe6643d4c2706b32c656fced1f11d8b5120dc39c
 
[ATTPdu.txt](https://github.com/user-attachments/files/19706889/ATTPdu.txt)

[BTMonWireshark.zip](https://github.com/user-attachments/files/19706908/BTMonWireshark.zip)

[HCIDump.log](https://github.com/user-attachments/files/19706909/HCIDump.log)

[BTMon_Serial_Logs.txt](https://github.com/user-attachments/files/19706913/BTMon_Serial_Logs.txt)

What can be the reason for this ??

hypfvieh, 2025-04-11:
So which side is using bluez-dbus/dbus-java?

I'm not sure if this issue is related to bluez-dbus or dbus-java.
This library is only a small layer providing the required interface files for accessing bluez using dbus (like a remote control).

For me it sounds more like a lower level problem, as you already pointed out that btmon also does not see everything.
If btmon is not able to get the complete data, bluez-dbus also can't.

As you are talking about Java 1.8, this ancient version of Java is not supported by bluez-dbus for quiet some time. Dbus-java does not support it since end of 2021. If this issue is anyhow related to bluez-dbus/dbus-java you should first try to use the latest versions of both libraries.

PavanNittala, 2025-04-12:
@hypfvieh  , Sorry for the confusion , i have all m observations written here , 
In summary ,  i can see data using btmon , which is reassembled by WireShark, 
I tried using Java 22 but the same result which is obvious since i dont see multiple writes on GattCharacteristic1

https://stackoverflow.com/questions/79570103/ble-bluetooth-not-receiving-all-fragments