hypfvieh/bluez-dbus #71: An error occurred when the connection was triggered

lunasaw, 2025-05-12:
org.freedesktop.dbus.exceptions.DBusExecutionException: le-connection-abort-by-local
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
        at org.freedesktop.dbus.errors.Error.getException(Error.java:116)
        at org.freedesktop.dbus.errors.Error.throwException(Error.java:143)
        at org.freedesktop.dbus.RemoteInvocationHandler.executeRemoteMethod(RemoteInvocationHandler.java:206)
        at org.freedesktop.dbus.RemoteInvocationHandler.invoke(RemoteInvocationHandler.java:89)
        at com.sun.proxy.$Proxy29.Connect(Unknown Source)

hypfvieh, 2025-05-12:
What is this? Some random stacktrace without any context, any sample code nothing.
How on earth should anyone help you when you are not willing to create useful tickets?!

hypfvieh, 2025-05-13:
This is a bluez error which is just forward by dbus to your client application:
https://github.com/bluez/bluez/blob/8154ddbdd5b2eaa5643391cca6bf2093288dabf2/client/bluetoothctl.rst#connect